maelstrom2001/ArcticFox #99: USB Charging Wismec rx 2/3

DrGeoGadget, 2017-01-30:
Charging of my Wismec RX2/3 is not fully completed. USB Charging stops, when one of the cells reaches 4.00 V. So the Cells will not be charged to 100% which is 4,3 V.

Greatings

maelstrom2001, 2017-01-30:
What BVO values are set to? What voltages displayed for other batteries when charging is stopped?

DrGeoGadget, 2017-01-30:
Hi

 

BVO Value ist set to 0,00V, the other batteries have between 3.95 to 3.99 V.

maelstrom2001, 2017-01-30:
Charge batteries with **external** charger, then leave them for 10..15 minutes and then insert them in the box. After that read batteries voltages on Info Screen.

> the other batteries have between 3.95 to 3.99 V

These values are displayed on RX2/3, or you measured them with voltmeter?
4,3 V for 100% charged battery is pretty high voltage.

DrGeoGadget, 2017-01-30:
When I charge the batteries in external charger the charger closes at 4.2 volt. The box shows these values on the Info screen. The values are displayed on the RX2/3 and I measured them with voltmeter.

maelstrom2001, 2017-01-30:
So, with all BVO set to 0.00 your RX2/3 show exactly the same voltages that external voltmeter? That's cool. Never saw such precise voltage reading on boxes. Generally, users have to adjust them with BVO.
OK, charging algorithm is designed so that when the voltage reaches 4.22 V on any battery, it stops. When charging stops, the battery voltage will drop a little (if battery is new) or more (if battery works for a long time), so normaly you should never get exactly 4.22 volts on cell right after charge stop.
In fact, the voltage measurement in a box is not accurate. So it is quite possible that the box is accidentally wrongly measured voltage, and decided to stop the charge sooner than it should.

DrGeoGadget, 2017-01-30:
ok thats fine, but what is the solution? With original firmware, there was no problem with usb charging. but I like yours more. is there anything I can try?

maelstrom2001, 2017-01-30:
Also check No Sleep option in Expert -> USB menu:
[image]

On boxes with 2 or more batteries setting this option to On is lead to slowly discharge while box is still connected to USB after charge end. Perhaps this explains that you find batteries charged to 4.00 volts. They can be discharged partially rather than not fully charged.
We have no 2/3, so we can not observe and say why it stops charging at 4.00 volts exactly.

DrGeoGadget, 2017-01-30:
OK thanks, I´m playing a little bit with different settings and BVO values. I tried a few minutes ago to charge batteries in the RX2/3 with a voltage above 4.1V each. It worked and i could charge until 4.19V. I´ll keep you informed about any news.

Deville505, 2017-02-01:
To the best of my knowledge the RX2/3 (as well as some of the other RS models) are horribly designed in respect of battery recharging via USB - I've heard remarks like "they load the 3 cells to 20%/40%/80% (i.e. unbalanced) but never full.
I acquired one myself 2 weeks ago and in my experience the recommendation in the manual to load the cells in an external loader is a requirement rather than just a recommendation.
I've used the original firmware as well as myEvic and ArcticFox, it indicates a problem of the board (some saved circuitry for battery management) rather than of any of the firmware versions.

Sanaki, 2017-02-02:
I never had any interest in USB charging my RX2/3 myself, nor did I attempt it. Having said that, I find it very unlikely that an RX2/3 would report accurate voltages for all three batteries. In my case my BVO settings came out to -0.08, +0.08, +0.01. My girlfriend's were similar from what I recall.

override187, 2017-02-16:
Well I can confirm that same error in my Tc200w, on stock fw it would charge until one cell reach 4.2v but on AF stops at 4.00v. Battery % was at 75, cells was at 3.85 4.00 4.00. I disconnected and reconnect charger and charging continued up to 4.20v.

closed 2017-05-15 by TBXin