maelstrom2001/ArcticFox #103: Display scrambles when using device monitor.

sandbagjohnny, 2017-01-31:
Hi, reporting intermittent screen artifacts:

Firmware: af_170129
Hardware: iStick Pico 75w, v1.00
Selected control mode: TFR 316L using PI regulation. No preheat.

Behavior: While running device monitor and changing PI-regulation values (using the device's interface), after editing the values the main display screen will jumble the on-screen characters. (ie- upside-down, backwards/mirrored. etc.). This happens very intermittently, so I have not been able to figure out how to reproduce the issue. It has not happened during any other interactions with the device. Removing and reinserting the battery restores correct behavior.

maelstrom2001, 2017-01-31:
Thanks, it's known issue, it seemed that it was fixed since 170121 build, but... no luck. Can't reproduce it, any help will be appreciated. Have Pico, and before fix in 170121 I can get screen artifacts even in standby mode with Device Monitor running. If we can determine the sequence of actions to reproduce this bug, it will be fine :)

sandbagjohnny, 2017-02-14:
So far I haven't been able to reproduce this issue with 170201... Just a followup note on the way the issue manifested though, don't know if it's related, but:

When I had this issue the first time, I thought maybe the FW had flashed incorrectly; when it scrambled, I figured "ok, I'll just reflash the stock fw and then go back to af.., maybe the  last flash didn't take properly".

When I went to go do this, the HW version info was found to have been set to an arbitrary number. So I reset the dataflash/reset the hw ver to 1.0, and reflashed to stock. fw. Then flashed af 170201 on.

It scrambled one more time. I pulled/replaced the cell and it was fine again. Issue has not happened since.

In either case (there were only two times this happened, total) the scrambling happened by holding + and fire, selecting PI menu, and then changing one of the values. 

I'm not sure if the length of time the monitor had been running was part of it or not(I had been changing PI values for about 10 -15 minutes)

The TL;DR is I haven't seen this again other than those two times. Hope this helps

closed 2017-05-15 by TBXin