maelstrom2001/ArcticFox #104: RTC adjustment question

Deville505, 2017-02-01:
I've played a bit with the RTC adjustment but I'm not really sure if I understood/ guessed correctly how it's supposed to work - my guess is that when you neither have a board with an X32 nor want to use LSL, you can manually adjust the clock speed value by increasing or decreasing it in the RTC menu or semi-automatically do so by adjusting the clock after some run time.
Doing the latter I'd expect to automatically change the speed value by calculating the difference between the new time and old time and  using the run time to change the old speed value to give a better estimate.
I was surprised that after doing several noticeable adjustments (in the 2-10 mins range after less than 24hours run time) the speed value shown was still the unchanged default value.
Is adjusting the clock NOT supposed to semi-automatically change the speed value, or is it just a glitch that the changed values are not show in the clock speed menu?

Hardware: RX 2/3, most recent ArcticFox installed, X32 and LSL both OFF.

maelstrom2001, 2017-02-01:
See #66 
Correction is manual, and anyway we suggesting to use Light Sleep Mode.

Deville505, 2017-02-01:
Thanks for the quick reply - I realized that the way I use the RX 2/3 it doesn't save much battery charge NOT to use LSL anyway so I switched it on again.

maelstrom2001, 2017-02-01:
> it doesn't save much battery charge NOT to use LSL anyway so I switched it on again

Exactly :) Noticed that all my 1 cell boxes (VTC Mini, Pico) with active LSL drops about 0,02 volts per 24 hour in standby. This is a very modest fee for accurate clock.

closed 2017-02-01 by maelstrom2001