maelstrom2001/ArcticFox #105: USB Charging Control ?

vapinggirl, 2017-02-02:
Hey there :)

Is it possible to Control usb charging current via Software ? A configurable limitation would be really great to be able to smoooooothly Charge overnight and save battery live and charging circuits

TBXin, 2017-02-02:
Hello,

Firmware is controlling charging process for all multi-cell devices (except devices with parallel battery connection).

Most of devices have 0.5A charge limit and only new devices like RX300 and Pico Dual have 1.5A limit.

Sanaki, 2017-02-02:
> Most of devices have 0.5A charge limit and only new devices like RX300 and Pico Dual have 1.5A limit.

spattinson, 2017-02-03:
I think OP wants configurable slower than normal charging? Most in device chargers are something like 0.5C anyway which is well within recommended limits for lithium batteries anyway. Probably no benefit in this option?

TBXin, 2017-02-03:
> Probably no benefit in this option?

Absolutely.

closed 2017-02-18 by TBXin