maelstrom2001/ArcticFox #106: support for eVic Primo

Cantpickrightusername, 2017-02-02:
Do you plan to add among suported devices eVic Primo?
U guys are doing great job. 
Thaks for all vapers and good luck.

TBXin, 2017-02-02:
Hello,

Yes, we will add support for this kind of devices, but at this point we are waiting for the official firmware release.

maelstrom2001, 2017-02-02:
Closed until official release.

ozomax1997, 2017-02-05:
I apologise

maelstrom2001, 2017-02-15:
Testers are welcome. Do not forget update NToolbox to the latest [nightly version](https://ci.appveyor.com/project/TBXin/NFirmwareEditor/branch/master/artifacts).
[af_170214.zip](https://github.com/maelstrom2001/ArcticFox/files/776645/af_170214.zip)

ozomax1997, 2017-02-18:
Tried out the firmware. Mod seems to het a little warmer on usb chargeing. Also is there supposed to be a screen off animation? Dont think my rx200s had it but not able to get my hands on it atm. Been using it for about a hour now and had no issues so far otherwise

maelstrom2001, 2017-02-18:
@ozomax1997 thanks for response! If it possible, can you compare box temperature with stock firmware while charging? I think that moderate heating is normal, especially that the charging current is 1.5 amps (on older boxes current is limited to 1.0 amp). But just in case it is useful to check :)
And yes, power off animation is implemented on all builds since 170210.

bjoernx, 2017-02-18:
I'm on it too now. on Linux (arch/Antergos). 
first I tried to upload with evic-usb (with --no-verify aprom option, dunno, no instructions found). thought I had bricked device, box remained totally unresponsive after that...
each time removing batteries and reconnecting helped, until then pc would not recognize no device. reflashed stock 4.05 with evic-usb, works! (pheewww.... there's my vape again, at last :D) 
managed to get a connection with w7/32bit in virtualbox (forgot to add me to vboxusers group first). unresponsive after flashing with the toolbox, again. same procedure with evic-usb (on the host). 
device was then recognized in the toolbox as some 0 firmware version thing in LDR (?) boot-mode ... don't remember, sorry. flashed with arctic, this time successful.
request: please add installation procedures with evic-usb on linux! :P 
only thing I have found so far is the single battery indicator, although there are two inside (similar to somthing I read with the Dual in another thread, maybe). but actually I really don't mind that, seeing two batteries makes little sense actually. otherwise really like what I see, will try this in more detail soon and come back at you, if something pops up.

nikosma, 2017-02-21:
Everything seems to be working properly, but i can see only one battery icon instead of two (anyway doesn't matter at all).
Excellent job!!! Thank you very much!!!!

bjoernx, 2017-02-21:
I second nikosma, the firmware is awesome! the more I use it, the more I like. :)

I found some annoyances with battery charging and depletion, but this will be more general. 
It's the lack of hysteresis both when batteries are depleted and in the fully charged status. This has the effect when charging, that there is no timeout. Lios should be charged for no longer than 4 hours max AFAIK. Charging should stop if either 100% are reached and voltages have not dropped below - say - 4V (fore example) - reset the flag then. 
On the other end I saw the device coming in and out of low battery status a few times when depleted, because the batts reload themselves a bit when empty. So there should be some hysteresis to pin the empty status as well. Also the low/empty-level maybe should be raised, since with the voltage drop voltage goes down as low as 2.8 V - I dunno if this is safe.

I guess I should open an extra Issue for all that, what do you think?

bjoernx, 2017-02-21:
Allright, forget what I said about voltage drops below 2.8 being dangerous. I mixed up with the specs of some 18650 I ripped from an old laptop pack a while ago (with < 1C current output), I wasn't aware that these high current things often have a much lower cut off voltage... XD the vtc5 I have here have 2.5 or 2 V, older ones have 3 (typically, afair).
Still, what I said about hysteresis would be nice to have... possibly. And the maximum charging time and avoiding trickle charging...? not sure.

akazuk, 2017-03-02:
Hi Guys. I've tried to update with NFirmwareEditor-v6.5.43 but it says " Opened firmware file is not suitable for the connected device. ". I've tried things like connecting/reconnecting usb, taking batteries out, etc. 
Sorry ... noob here. Can anyone help ? Also in the field Device Name it says Unknown device. 

My firmware versions is 4.05 hardware version 1.0

TBXin, 2017-03-02:
@akazuk 

Hello,

Are you trying to install nightly build of the ArcticFox firmware that was attached in this issue above?

akazuk, 2017-03-02:
Yes, lol idk what happened but it worked now. Thanks anyway :)

sarathmenon, 2017-03-09:
Is there a more recent firmware? I tried the one linked in here, but on flashing, the device won't boot. I had to restore stock to get it working again. I am using stock windows 7 on a virtualbox VM, with the usb device added to the VM.

maelstrom2001, 2017-03-09:
The most recent firmware is located [here](https://github.com/maelstrom2001/ArcticFox/wiki/Changelog), also there you can find the link to nightly NToolbox which is required to install the firmware.

Please take a look on https://github.com/maelstrom2001/ArcticFox/issues/106#issuecomment-280884255
If you're using VM, box after flashing can stuck in LDROM boot mode (see status in Firmware Updater). Just flash the firmware .bin again (if you sure that it is not corrupted) or switch the box to APROM boot mode manually:
[image]

sarathmenon, 2017-03-11:
@maelstrom2001 Thanks for the comment. I did read through all the comments in here, but somehow missed #106. I am using the nightly NToolbox, but after flashing the latest firmware, I still get the same thing. The screen stays off, and NToolbox does not recognize the device to allow me to change the boot mode or flash the firmware again. I am on OSX, which shouldn't matter that much in the grand scheme of things. runnning NToolbox with wine gives me the same thing. I have a linux host that I might be able to retry this on.

espezetka, 2017-03-18:
Hi guys, at first let me please say warm Thank you for your work for us. 
And if i may ask you. How does it look with full support for Primo? Any eta of release? 
Now i am using v. 170222 and everything is working fine, only one thing what is not actually acurate is battery meter. The Primo is charging only to 4.13V each battery(LG HG2 or VTC4, paired batteries) . I haven't it measured by voltmeter so it might be only a number. 
Thank you again for answer and have a nice day. ^_^

maelstrom2001, 2017-03-18:
@espezetka thank you!
Developing is in the active phase, ATM we can't say when there will be a release, too much work remains to be done. We looking for feedback on nightly builds, 'cause we have limited number of supported boxes and we can not test everything physically :)
If you want, you can test attached build. Charging routine was modified slightly, need to test it and check that the charging on this build goes through all stages correctly. Any help in testing and responses are highly appreciated.

[af_170318.zip](https://github.com/maelstrom2001/ArcticFox/files/852716/af_170318.zip)

espezetka, 2017-03-18:
Thank you. I will charge now and once the charging will be complete i will report.  I will try it twice with mentioned batteries and let you know

espezetka, 2017-03-18:
Ok, LG hg2 now ended  charging.  The highest voltage i got was 4.15 and 4.14 volts. Then, still plugged into wall adapter, voltage dropped to 4.12 and 4.11 volts without any firing etc. Tomorrow i will try charging with VTC4 and with device monitor recording and will upload a csv file here.

maelstrom2001, 2017-03-18:
> with device monitor recording and will upload a csv file here

It will be very useful :+1: suspect that while charging there is spikes in battery measurements occasionally, i. e. wrong measured and overestimated voltage cuts charging off. Monitor data at the end of charge will slightly clarify the situation. Will see :) thank you!

espezetka, 2017-03-19:
Lines 84-87 there you can see the voltage drop when charging. Device was still plugged in and nothing happened since the drop. Now i am tryinng VTC4 so i will upload a next file to this comment once the charging is complete.
[primo lg hg2.zip](https://github.com/maelstrom2001/ArcticFox/files/853067/primo.lg.hg2.zip)

espezetka, 2017-03-21:
Well so i changed to original firmware from joyetech and here is the result so i can say that the hardware is all right. 
[image]

Other devices with ArcticFox will charge to 4.19 without any problems, but theese are a single or built in batteries. Istick pico and ipower 80w.

maelstrom2001, 2017-03-21:
@espezetka thanks for report, working on it :)

espezetka, 2017-03-21:
For sure mate :)

maelstrom2001, 2017-03-24:
@espezetka can you test attached build please? Make some changes in charging routine, maybe "4.12 volts bug" will disappear :) Just in case, please do not leave mod overnight and monitor it time at time to avoid overcharging. It should stop and turn screen off at 4.19..4.23 volts depending on the voltages difference.

[af_170324.zip](https://github.com/maelstrom2001/ArcticFox/files/868961/af_170324.zip)

espezetka, 2017-03-24:
Ok no problem, i will take a look on it tomorrow. 
Now my batteries are on 3.6V both so to 4.something V i will be on wall adaptor and then i will swich to pc  USB and start recording trhu devices monitor and Post a csv file here :)

espezetka, 2017-03-25:
So dear Mr. Maelstrom2001 here are results after charging.
4.19 and 4.21 V for me looks good but i have to point out the board temperature. From 44 celsius on older build to 36 celsius on new one? Nice :) 
For me, this version is much better than was 170318. 
Batteries are full and the PCB is not exposed to high temperature while charging. What else to wish?
 TFR with same curve as i have used on older version, seems to work better on new version, but it might be only my opinion because i do not know if you was doing something with TFR functions too.
For me, this version is just a top notch for Primo :+1: :100: 
[LGHG2 new.zip](https://github.com/maelstrom2001/ArcticFox/files/869892/LGHG2.new.zip)

maelstrom2001, 2017-03-25:
@espezetka thanks a lot for testing and response!
I see in attached CSV that charging was started with 4.13 volts, right? Can I ask you to make the last test? :)
Please discharge batteries to approx. 4.0 volts and put box on charge. No need to record CSV this time, just make sure that charging will continue when voltages will reach 4.12 volts (transition to 300 mA charging current, switching to the last charging stage with balancing). Thank you!

espezetka, 2017-03-25:
Ok i will swich atty to be able to use higher wattage and faster Fischerem and will report soon. Will use wall adaptor 2.1A

espezetka, 2017-03-25:
Discharged to 4.01V both, charging slowed down since 4.12/4.13V and continued to final 4.19/4.20V and then screen shut off. So far so good for me. :+1: 
[image]

maelstrom2001, 2017-03-25:
[image]

espezetka, 2017-03-25:
[image]
:1st_place_medal:

andreivoinea93, 2017-03-25:
I just tried this and it fried my evic. -.- Seriously. Smells burnt and won't start

espezetka, 2017-03-25:
Any more info? Battery type?

andreivoinea93, 2017-03-25:
VTC6 charged at 80% (as shown by Smok Alien)

espezetka, 2017-03-25:
Were the batteries paired? Same age, cycles etc? Was Evic hot before it died?

andreivoinea93, 2017-03-25:
The batteries never seen different mods, they were always together in the same mod, either my Primo or my Alien. Board temp during charging first was 50C and then cool off to 47. I took out the charger, plugged it back in, the screen started flashing like it was in a bootloop and then it died.

espezetka, 2017-03-25:
Did you tried to insert batteries again without charging?

andreivoinea93, 2017-03-25:
sure, nothing happens.

espezetka, 2017-03-25:
So let's wait till Mr. Maelstrom2001 take a look on this thread. Maybe it was HW problem.

andreivoinea93, 2017-03-25:
I doubt. It would be the too much of a coincidence.

espezetka, 2017-03-25:
Well try this. Download official FW from joyetech page, try to flash it into mod trhu official updater. I know about similar issue but there was no smell of burning. Maybe it helps

andreivoinea93, 2017-03-25:
It would be helpful to know if based on the software change the problem could be a blown fuse, so I can fix that myself. Otherwise I have to return it

andreivoinea93, 2017-03-25:
I'm running linux so I can use lsusb to check if any device is alive on USB. It shows nothing. It's dead

espezetka, 2017-03-25:
give it a try, if you will be able to put original fw back you can return it without breaking a warranty

andreivoinea93, 2017-03-25:
Great! Joyetech website is down now, rofl.

I'll give it a try but there is no way it's gonna work :)

TBXin, 2017-03-25:
@andreivoinea93 did you try connect device to PC without batteries and holding "plus" button?

1. Remove batteries
2. Press "plus" button and hold
3. Connect to PC
4. Release "plus" button

After this try to find device via libusb. If you find it - so, you can install any firmware.

espezetka, 2017-03-25:
[joyetech.zip](https://github.com/maelstrom2001/ArcticFox/files/870176/joyetech.zip)
here is the original FW and updater for primo

andreivoinea93, 2017-03-25:
Nope, no response with the plus button. Needless to say, joyetech updater doesn't see a device.

espezetka, 2017-03-25:
This is really srtrange... Mine works fine, now charging again just for try

maelstrom2001, 2017-03-25:
> I took out the charger, plugged it back in, the screen started flashing like it was in a bootloop and then it died.

> joyetech updater doesn't see a device.

Looks like MCU or its voltage regulator died on USB connection, really HW fault. Sad, but nothing supernatural. Remember how many were RX2/3 dead - black displays, burnt DC/DC converters etc.
Posted yesterday on Russian forum: http://www.ecigtalk.ru/forum/f17/t141141-post10289363.html#post10289363
"Burned. It was the first my device which supports USB charging, and seems exactly that was the problem".

espezetka, 2017-03-26:
Few chargings later : still charging fine, without any problem. Device is still around 35°C when plugged into wall adaptor. Once it reaches 4.12/4.13 it slows down. 

Just reporting :)

maelstrom2001, 2017-03-26:
@espezetka Do you know what is the most amusing? This is what the builds 170324 and 170318 are identical on MCU side, checked with disassembler :) I have a suspicion that charging is depend on batteries voltages difference. Now you have perfectly balanced cells, and charging works very smoothly. But the same build refused to work correctly on RX300 with a slightly imbalanced cells. And finally: charging routines in stock Primo and RX300 firmwares are identical. At the moment I am in a slight bewilderment :)

espezetka, 2017-03-26:
About batteries, every monday i am putting them into external charger so maybe this is the reason for balanced cells. I recomend this to everyone, your mod might have balancer (primo has)  but external charger is a basic device for vaper. 

Maybe died primo was just a bad luck, it just happens. Sensitus was not first or last with hardware failure i think.

OlafKroll, 2017-03-29:
170318 works fine on my Primo

RX300N, 2017-04-06:
170331 - latest I know about - works fine both for Primo and RX300

Pete103, 2017-04-14:
170331 works fine on my Primo too with one issue being the clock being far from accurate. My impression is Primo has an RTC chip, is this correct?

skorpyx, 2017-04-16:
still reports "atomizer short error" if you reinstall back to official firmware so everything is fine. What is the solution?(170331) Thanks

maelstrom2001, 2017-04-16:
@skorpyx please provide more information about your setup. Have Primo (early revision) but can not reproduce and confirm this issue on any of my atties.

[20 more replies]

closed 2017-05-08 by maelstrom2001