maelstrom2001/ArcticFox #107: Support for iStick Power/iPower new 1.03 firmware and 100 watt mod

iJo09, 2017-02-02:
I see there's no support for the new iPower 1.03 firmware.. can you please add it and make the battery do 100 watt (like with VTCFont's patch)?

TBXin, 2017-02-02:
??? 
This is a custom firmware repository. And our firmware is supporting iPower devices.

iJo09, 2017-02-02:
Ok so can you change the maximum power output to 100w on iPower? As I said there's a patch for the original fw and it doesn't do nothing bad

TBXin, 2017-02-02:
Single battery can't give you 100wats, so, patch is useless.

iJo09, 2017-02-02:
So not even 80.. plus it's an integrated battery and certainly isn't at the level of a high-end vape battery

maelstrom2001, 2017-02-02:
Maximum power limit for all devices is set to value from recent original firmwares. When Eleaf will increase it to 100 W for iPower - no problem.

closed 2017-02-02 by maelstrom2001