maelstrom2001/ArcticFox #109: Limit hit time option?

nextcode4u, 2017-02-02:
A feature that I personally believe would be nice (if its not already implemented and I missed it.) 
Is to set a time limit on hits for profiles such a limi to 3 seconds per hit. This could help misfires, along; if you let someone else hit your vape this rules out the possibility they would burn your coil.

Technetium1, 2017-02-02:
I second this as a very interesting feature!

TBXin, 2017-02-02:
You want decrease on increase delay for interpreting multi-clicks?

nextcode4u, 2017-02-02:
That could be something to add as well, but I meaning if you hold the fire button you could limit how long it fires before you press the button again 
ex:) when I take a hit and hold the fire button it will only fire for 3 seconds (if I set that time) even if im continuing to hold the fire button until i release and press the fire button again it will then fire for that set time.

TBXin, 2017-02-02:
[image]

Oh, so you are looking for this feature, and it is configurable only from Toolbox.

nextcode4u, 2017-02-02:
SWEEEEET needed that! thanks!

closed 2017-02-03 by maelstrom2001