maelstrom2001/ArcticFox #112: iStick Pico Battery Charging Issue

barsuc, 2017-02-04:
When i'm charging my battery by USB there is something weird heppen. When i plug USB with charger to the mod, battery level rise qyickly by about 4%. When i unplug device, it drop by this 4%. Will it be issue with software or battery?

TBXin, 2017-02-04:
Single battery devices like iStick Pico has hardware implemented charging, firmware can't control it.
Try another battery.

barsuc, 2017-02-04:
Ok, thanks for the answer ;)

maelstrom2001, 2017-02-04:
@barsuc it's normal. Percentage calculation is based on battery voltage, which increases when you plug USB in and decreases when you unplug it. This is especially noticeable in the middle of the discharge curve, where a very small difference in voltage results in a significant difference in the percentage.

closed 2017-02-04 by maelstrom2001