maelstrom2001/ArcticFox #116: vtc basic support

Wietjuh, 2017-02-04:
A friend of mine has the VTC Basic and i think there is a different internal battery.

Has anybody tried it yet for this device? Can someone confirm compatability?

Thanks in advance

Wietjuh

TBXin, 2017-02-04:
This is not enough?

[image]

maelstrom2001, 2017-02-04:
Battery discharge profile should be set to GEN, Basic has the same battery type as in Cuboid Mini, eGrip, VTwo etc. Different capacity, but the type and discharge curve are identical.

Wietjuh, 2017-02-04:
Tnx Maelstrom, 

And to TBXin, last time i tried with a december build the device was decharging faster than with regular software and the voltage wasn't correct. It read in the live monitor just 3.7v even fully charged. So i will try later tonight for the newest build..

TBXin, 2017-02-04:
@Wietjuh perhaps it was a bug that we are fixed later.
At this point eVic Basic shoulde use correct discharge profile in our firmware.

Wietjuh, 2017-02-05:
It reads the correct voltage now and it works like a charm! Thanks!

closed 2017-02-05 by TBXin