maelstrom2001/ArcticFox #117: Q: Display issue

WatIsGithub, 2017-02-05:
Hello, yesterday i upgraded the firmware by the instructions on the official web, with latest toolkit and to latest version on my RX2/3.

Everything was ok, but after recharging my batteries it has written number 6 on all the free space on the profile selection screen above the profile name, and even for a while after selecting one i wanted. I had active only 2 profiles done in toolkit, after few seconds it disappeared, but everytime it was there for a while. Today it was working, no numbers, so i went to work, but after opening the back cover and placing it back all the non active profiles appeared back as active, with no. 6 all over the top part of the display.

After another 2 hours, now, the display is lit, as i can see it while looking at it from the top, but no text is showing, no info. Just like when it is in sleep mode. Mod is working, i can vape, change the profile by memory, but cannot see a single pixel white. Is there some button combination i shall press to get it back? Is it stealh mode i activated by accident? Fire+down doesn't work as in orig fw. Thanks for info

maelstrom2001, 2017-02-05:
Try to flash stock firmware. BTW, is RX you bought from last batches? Saw some info that latest 2/3 has faulty displays in mass.

WatIsGithub, 2017-02-05:
Well, bought it ... not even month ago. Don't scare me! :( I will try to reflash it to the AF once again without changing a thing, if it happends again, stock will be prepared in my laptop. Going for free testing this evening and this would mess up my joy of it, since i am not sure what temp etc i have set there.
I hope it will turn on if i reflash it, don't wanna wait 2-3weeks before the techs in the store inspect it to return the money if it is faulty display :(

Before i get to that, in worst case scenario, what similar mod would you recommend? 3 cell, up to 200/250W? AF supported and without this kind of HW issues

maelstrom2001, 2017-02-05:
> Don't scare me! :( 

If you want to be scared even more see #82 :)

Well, I think that RX200S has less issues and it is suitable to your needs.

WatIsGithub, 2017-02-05:
After reflashing to stock, back to AF, back to stock, trying it without cells and with it i came to conclusion ... Dead display :( But those ~12 hours with AF were amazing ... so i am looking for something with support of your fw or similar, but only 3cell mods with this kind of shape are leather covered DNA250 mods- Triade,Vape Finder and Hcigar, where i dislike the leather case... Is RX200S more reliable? Any known issues?
Thanks for help anyway, i am glad there is such support on custom fw which is so amazing :)

maelstrom2001, 2017-02-05:
> Dead display :(

My condolences. The problem is a very common, [for example](https://www.reddit.com/r/electronic_cigarette/comments/5mla7t/wismec_reuleaux_rx23_blank_screen_problem/).

> Is RX200S more reliable in this kind of issues

Probably, yes. It seems that RX200S users have much less of problems than RX2/3. Suggesting you to Google and compare results for both boxes fault reports.

> Thanks for help anyway

Thank you, and hope you will lucky with new box!

WatIsGithub, 2017-02-05:
So it´s settled, i will get RX200S tomorrow as a secondary mod, since i was quite happy with it until today and it suited my needs well, also it is much cheaper compared to other mods and i love the rough design.
Thanks for help, i will test your fw fully tomorrow :)
Me love you long time :D

closed 2017-02-05 by WatIsGithub