maelstrom2001/ArcticFox #125: Creating custom discharge curves

Sanaki, 2017-02-08:
I'm picking up some Sony VTC5As and was hoping to input the curve into my mod. I have mooch's discharge curves available from [here](https://www.e-cigarette-forum.com/forum/threads/sony-vtc5a-2500mah-18650-bench-test-results-a-fantastic-25a-battery.746719/) but at the moment I'm trying to figure out exactly what the best practice for inputting them would be. I would assume the constant discharge curve will come in low, since we generally read our battery percentage when the mod is resting. Does anyone have thoughts? Attempting to compare preset curves to the graphs I could find didn't lead me anywhere firm.

maelstrom2001, 2017-02-08:
https://github.com/maelstrom2001/ArcticFox/issues/96#issuecomment-275926428

Sanaki, 2017-02-08:
I searched, I failed. Thank you.

closed 2017-02-12 by TBXin