maelstrom2001/ArcticFox #127: [ISSUE] Resistance lock TC mode

ceinro, 2017-02-09:
Hello,

First of all i have to congratulate on this magnificent "update" for my Pico 75w.

Soon i started using TC and the resistance never stays locked even if the little padlock is there and the option is ticked from the NToolbox.  The resistance will " lock " on its own when the mod is not used for a couple of minutes. The new resistance is of course higher ( using Ni ) and TC doesn't work how it's supposed to. If i set myself manually the resistance again and again it's working ok but i want it to stay locked and saved to a profile. How can this issue be fixed ? 

Also, what is PI-Regulator? I have not yet enabled this option but i am curious. 

Thank you.

TBXin, 2017-02-10:
Hello,

Actually you can't see "locked" resistance on the screen. Resistance will flow during heating coil, we calling this "Life resistance", but firmware are used locked resistance to calculate temperature.

PI-Regulator is an advanced TC setup tool. When it is turn on - firmware will use alternative algorithm to maintain a predetermined temperature. "PI" is a special case of "PID" regulator, but without "D" 😄 
PID means proportional–integral–derivative controller (PID controller).

You can read some infos here: https://en.wikipedia.org/wiki/PID_controller

> resistance locked shows wrong resistance but if i fire one time it goes back on
Yep, resistance can 'stick' for sometime, but when you press Fire button it will immediately recalculate and you will see correct value on the screen.

ceinro, 2017-02-10:
Exacly this is the problem. When smart profile is activated the resistance will stick to whatever the resistance is in that moment. If i disable smart profile on 1 fire click resistance is back to whatevere i fixed it initially or what i set up in profile. How can i keep smart profile on and dont mess up the resistance. Thanks

TBXin, 2017-02-10:
@ceinro may be you've created a few profiles for single atomizer with similar saved resistance and firmware just choosing wrong profile?

ceinro, 2017-02-10:
No, it keeps the profile selected but alters the resistance. Maybe a video will be self explanatory?

TBXin, 2017-02-10:
@ceinro Would be great.

ceinro, 2017-02-10:
@TBXin Here is the video : [https://youtu.be/CtOEylw8PKI](url)

I left only one profile so you can see the problem.

ceinro, 2017-02-10:
I have found out that enabling Ed New from Smart settings resolve my issue as in i click if i want to update the profile or not.

TBXin, 2017-02-10:
@ceinro I watched the video, and that's what i noticed:

1. Your saved resistance 0.183 is lower than real 0.193
2. When you are disconnecting/connection atomizer - you are doing it's very slowly and when coil is hot;

How smart works?
When you are attaching a new atomizer or existed one - firmware is trying to find profile with similar resistance +/- 10%.
In your case (1 profile with saved 0.183 ohms) there are no suitable profile for atomizer with resistance 0.229 (0.229 is much higher than 10% from 0.183) and you see "Assign profile" dialog, when you are pressing fire in this dialog - your are changing saved resistance (0.183) to actual (0.229).

I must admit following things:
1. Try to attach atomizer faster - this is good practice and helps devices to calculate  resistance with more accuracy;
2. Do not reattach atomizers with a hot coils, because materials like Ni200 can change their resistance very much after a few puffs.

ceinro, 2017-02-10:
So in few words if i want to use smart auto and TC and disconnect de atty i have to wait till it cools down. 
Also it forces me to set the live resistance as locked one from time to time. It is not a problem but it is inconvenient.
I will live smart off and manually select the profile when changing attys. It's not that hard ...

Thanks for explanations !

closed 2017-02-10 by ceinro

TBXin, 2017-02-10:
>Thanks for explanations !

Always happy to help :)

> Also it forces me to set the live resistance as locked one from time to time. It is not a problem but it is inconvenient.

Hm, why? When you are using TC mode, usually you are locking resistance once, when you installed attys, when coils are "cold" and then you are just vaping ^^

Also, if you got only one attys - smart mode is useless, you can simply turn it off.

ceinro, 2017-02-10:
I have 3 attys and keep changing them often. 
I will test more and if i come with other issues i will explain but for now i am pleased with my pico.