maelstrom2001/ArcticFox #133: Couple of small issues (EVIC VTC Dual)

pi-xel, 2017-02-11:
Hey,

First of all, thanks for the great work, really enhances usability of my device. However, I've noticed a couple of small issues:

- When I enter the edit profile screen and exit thereafter, the device does not return to the edited profile, but to the first, which is quite annoying.
- After using the device with the configuration software, it cannot be completely turned off anymore (after turning off battery statistics are shown). Have to remove battery such that it works again.
- Have some occasional freezes, mostly when entering main menu. Also here I have to remove battery (lucky me that I use a removable battery device :) )

Dunno whether some of the described behaviors (besides the freezes) are intended. If so, it would be nice to evaluate why it is such that and whether this can be changed. Thanks in advance.

Best regards,
Phil

maelstrom2001, 2017-02-11:
It seems that something goes wrong. Reset Data Flash with Firmware Updater and reinstall firmware. This should eliminate freezes and incorrect behavior.

pi-xel, 2017-02-11:
Thanks for the quick response. Did the reset and reflashed. After flashing the freeze happened again when entering main menu and the battery screen on switching off is still there. Removing and re-inserting battery fixed both for now. Will report if it occurs again

Regarding the profile switching: It seems that smart detection is the cause of my problem, as it seems that smart detection is also run if the profile screen is entered and exited again, throwing me back to profile 1 if profile 1 coil is installed. Is this desired behavior? Can you tell me when smart detection is done?

maelstrom2001, 2017-02-11:
> Is this desired behavior?

Yes, it is by design. A little trick: if you're closing menu by selecting Exit item, atomizer will be re-detected and smart selection will be called. But if you'll press Fire and + button (such as for entering menu) instead, atomizer will not be re-detected, menu will be closed only.

pi-xel, 2017-02-11:
Ok, thanks for the info. But wouldn't it be a better idea to do detection only on boot and in the case that the current atomizer is removed (i.e. resistance 0 ohm)? Or at least give this option? :)

maelstrom2001, 2017-02-11:
Atomizer detection is tied to a pretty large amount of settings. For example, when uploading the configuration from NToolbox we have to do something like "hot restart". The same goes for some of options in menu. Maybe we'll think up something :)

pi-xel, 2017-02-11:
I see, but smart detection (not atomizer detection) could be disabled for these occasions, right? I mean, after all it can also be disabled entirely. I think this behavior can even be sometimes "dangerous" if you enter for instance a menu when the coil is hot and after exiting the menu it thinks that another coil is installed. 

For now I will simply disable smart, but thanks again for your help anyway.

And another question: Will ArcticFox at some point become open source?

maelstrom2001, 2017-02-11:
> after exiting the menu it thinks that another coil is installed

Yes, it sounds reasonable, perhaps will be useful to add a temporary blocking of Smart selection on exit from menu. It will not be easy to implement, but it's entirely logical. Thanks for suggestion :)

> another question

Yes, [some day](https://github.com/maelstrom2001/ArcticFox/issues/83#issuecomment-274794165)

pi-xel, 2017-02-11:
Alright, thanks for your help and have a nice evening.

pi-xel, 2017-02-11:
I've just nailed down the steps to reproduce my error of not properly shutting down (showing battery charging screen on shutdown) and of freezing in menu. Steps to reproduce:

- Disable USB charging
- Connect device to PC and open configuration editor (dunno whether something needs to be uploaded to reproduce or whether it's already sufficient to connect the device to a USB charger)
- Disconnect device from USB
- Shutdown device. It should not turn black but show battery charging screen instead
- Start device again. Try to open menu using "fire" + "+"
- Try to move to another menu item. Device should freeze, only way out is removing battery.

Please tell me whether you can reproduce my issue.

maelstrom2001, 2017-02-11:
> - Disconnect device from USB
> - Shutdown device. It should not turn black but show battery charging screen instead

Disconnected device is still show charging, right? In both cases, when USB charging is enabled and if it's disabled?

pi-xel, 2017-02-11:
Yes, it shows charging after disconnect, only if USB charging is disabled. And then it also does not go away, but returns to the charging screen each time the device is turned off. 

If USB charging is enabled this does not occur, i.e. the device turns black on disconnect and power off. 

And the freeze also only occurs if USB charging is disabled.

maelstrom2001, 2017-02-11:
Thank you! And the last question: is this bug present in both configurations, with 1 and 2 batteries? Or only in one of these?

pi-xel, 2017-02-11:
Will check that tomorrow, right now I am using it in 1 battery mode.

maelstrom2001, 2017-02-11:
Yep, suspicions were confirmed. I think you'll have no problems in 2 cells configuration. Will look what we can do with "single battery bug", and in day or 2 will ask you to test.

pi-xel, 2017-02-11:
Ok I can confirm two things:

- bug also occurs on USB charger, no need to connect to PC
- indeed with 2 batteries the bug does not occur, with 1 battery charging screen on power off and menu freeze

maelstrom2001, 2017-02-11:
Thanks, working on it :)

maelstrom2001, 2017-02-12:
Unfortunately, we have no VTC Dual to test, so please try to flash attached build. Please answer, is the bug still reproduceable.
[af_170212.zip](https://github.com/maelstrom2001/ArcticFox/files/769346/af_170212.zip)

pi-xel, 2017-02-12:
Both bugs (menu freeze and charging screen on power off) are gone with attached build. Thanks for the quick fix :)

maelstrom2001, 2017-02-12:
Thanks a lot for bugreport, testing and response!

pi-xel, 2017-02-12:
You're welcome, happy to help (even more as I am therewith also helping myself :) )

closed 2017-05-08 by maelstrom2001