maelstrom2001/ArcticFox #134: Preheat Delay with Curves?

mumpen, 2017-02-11:
Love this firmware, but why can't I enable preheat delay with curves?
A simple delay would be nice. Even better would be the option to ignore all values above 100%!

TBXin, 2017-02-11:
Hello,

We decided that it would be superfluous. When you are editing curve you can slightly reduce last few points, depending on your average puff time.

I mean something like this one:

[image]

At the beginning you get some boost, but when coil is already hot no need to give so much power, so it can be reduced to stabilize temperature.

Yep, this way of setup is more complex than just regular delay, but this way it can give you a much more possibilities.

mumpen, 2017-02-11:
Thanks for the reply!
But that's not what I meant, my power curve already looks similar to your example. 
If I take another hit a few seconds after the first, my heavy coils are still hot and the preheat from the curve is too much. I would rather like the power to either stay at 100% or even go down like in the curve.

maelstrom2001, 2017-02-11:
We think that precise power regulation, as it implemented with Curves, is more suitable for lightweight coils. If you are using heavy coils, the better way is to use simplified preheat type (with dead time settings), just power factor and time. Because of the high inertia of the precise power control is meaningless IMHO.

mumpen, 2017-02-11:
If you say so... I'll have to go back to the myevic custom firmware then :(

closed 2017-02-11 by maelstrom2001