maelstrom2001/ArcticFox #135: [VTC Dual - GENERIC]

guglielmoTaro, 2017-02-12:
Due to EXTREME DISCREPANCY between REAL and READ coil value i'd like to have the SHUNT CORRECTION parameter unlocked (<85 - 115>)

pi-xel, 2017-02-13:
Hey guglielmoTaro,

I also own the VTC Dual and my ohm meter seems to work rather accurately, maybe a drift of 1 to 2 percent, but not more and certainly within the +- 15% range of the shunt correction.

Do you have another mod or a resistance meter to compare to or are you only relating to the calculations on steamengine? If the latter is the case, then I would suggest to account for the leg lengths of your coils accurately and to check whether your coils are indeed tightly screwed to the base.

If you see a lower resistance on another mod or on the ohm meter then the 510 connector of the Dual could be the problem, because it is rather deep, such that "shorter" tanks may have connection problems there. On some tanks the length of the 510 connector can be adjusted, which ones are you using?

If all of these are not sources of your problem then I would think that your Dual is indeed broken in some way (e.g. has a loose connection internally) and consequently shunt correction would not be the measure to take, but rather repair or replacement.

Best regards,
Phil

guglielmoTaro, 2017-02-13:
I was getting strange values SOI tried different tanks and it's indeed sorta working... all of sudden it also gets 0ohms so I'll be going to have it replaced! Many thanks!

closed 2017-02-13 by guglielmoTaro