maelstrom2001/ArcticFox #136: request: Flashlight via screen

Overclick, 2017-02-14:
В устройстве очень сильно не хватает функции фонарика. Будьте добры, добавьте такую функцию на мультиклик, чтоб подсвечивала все пикселя экрана на максимальной яркости. Расширить функционал можно настраиваемым таймером, ну или хоть фиксировано 5-10 секунд, думаю для крайних нужд будет достаточно.
Полноценным такой фонарик не станет, но я своим пико и так порой в темном подъезде подсвечиваю, думаю не я один ))

wackydude1234, 2017-02-14:
My pico's screen already has burn in from regular use, I doubt this would be a good idea.
Pico экран уже горят в от регулярного использования, я сомневаюсь, что это было бы хорошей идеей.

Overclick, 2017-02-14:
to wackydude1234
You may set up dimmer to 0% for general use. I did that for my one after your warning, thanks. But I need the flashlight sometimes. Nothing going to happen for few seconds of screen burning, just lets limit that function for short time and board temperature. 3 sec if board < 45'C for example.

wackydude1234, 2017-02-14:
I've already done that but thanks :)

closed 2017-02-28 by TBXin