maelstrom2001/ArcticFox #138: iStick Pico TC

buddermilch, 2017-02-15:
Hey there, first of all thank you for creating such a great firmware. I'm using it in my rx2/3 without any problems. 

But it seems like I can't get TC to work properly on my iStick Pico. I first thought it might be the Clapton coil I was using so I installed a simple single coil SS304 spaced. Switched TC to SS304 and did a dry burn test, and that's when I first noticed it. The cotton got burned pretty badly. 

I'll make this short, whatever I do Temperature will always be displayed between 120-150 degrees Celsius. Even when I set wattage to 50 watts, which hits way to hard so there is actually way more temperature than displayed. The temperature also jumps from 150 to 80 back to 130 and so on.

I tried to enable x32 nothing changed, I used TCR mode set to 105 and even changed the TFR? Curve with NFtoolbox, nothing changed its behaviour.
The same atomizer works flawlessly on my rx2/3.
I also flashed the firmware again, to see if it changes anything, but it didn't.  

I actually can't help it anymore, is it an error on my end ? 

By the way, I flashed the firmware on both devices using a Windows VM on arch. Maybe that's causing the problem ? But as I said it's working on my rx2/3.

I hope you guys can help me, thanks in advance.

joeblowma, 2017-02-16:
TBH it sounds to me like you haven't got the resistance set to the correct value on the pico for your atomizer. Basically, the temp displayed while firing is calculated from the current measured resistance of the coil against the set "cold" resistance of the coil, with the TCR/curve describing how much added resistance equals what temp increase. If it's set too high it will always be trying to heat it too much to raise the resistance further to meet the heated resistance value described by TCR or curve. Cranking up the wattage just makes it feed more power to try to meet those conditions and burns your cotton faster if it's aiming for a resistance it can never achieve due to too high of a "cold" resistance.

That said, I had similar problems initially with TC on my pico and had to thoroughly clean the 510 connector on both the device and atomizer, a couple of very blackened cotton swabs and a little extra tighten down on the atomizer fixed it. What tipped me off to this was watching live resistance set on a display line as I tightened the atomizer down (AF is so great for things like this). I have used spaced, unspaced, clapton and twisted coils in 3 RTA and 2 RDA all with ss 316L in TC mode on my pico and have always been able to get sane temp settings/readings so long as my ato is snug and the resistance set correctly for the coil.

I've read in various places that some atomizers simply don't play well with the 510 connector on different devices (works properly on one kind, and not the other) due to centre pin depth. Sometimes you can adjust the pin on the ato, or get an intermediate heat sink with an adjustable pin to correct the issue. There is also always the possibility of a defect in the device, if it's new to you.

I suppose a couple things to try (if you haven't already) to see if its an AF problem... does it work properly on official firmware but not AF? Does that atomizer work at sane wattage levels in VW mode on pico? Have you tried reflashing the latest non-nightly _and_ wiping the settings (toolbox-updater-dataflash tab-reset dataflash)?

Good luck, hope something I said here can help.

buddermilch, 2017-02-16:
Hey thank your for your kind replay. 
I will quickly reply to your last part, I don't have much time right now. But I will try your above steps later when I'm home again. 

Yes it did work pretty flawlessly on stock eleaf FW.
Yes it work even when set to 75Watts if you mean that.
No I did not dataflash, but I reset the settings in the configurator before flashing the FW again. 

And yes I also did thigten the screws as I always do, the resistance stays at a steady value. Which I set with the atomizer at room temperature, I did that a few times to see if that's the problem. Although I did not lock the resistance, I never did that. But I will try it once I'm home and check the screws and the 510 connection. 

Thank you.

TBXin, 2017-02-16:
Hello,

> Although I did not lock the resistance, I never did that

It's a bad decision. Correct locked resistance is a very important part of the TC mode.
I'll try to explain ^^
Temperature is calculating from live/actual resistance of the coil, but device can't calculate it without any previous "points". Under points i mean pair of the temperature and resistance. First point is always "room temperature" (68F / 20C) and some resistance, if you locked correct resistance - it will be here, otherwise firmware will think that actual resistance equals to the room temp and this will cause problems if your coil is hot at this moment ^^

joeblowma, 2017-02-16:
> Yes it work even when set to 75Watts if you mean that.
Not so much, I mean I fire a .670 ohm coil at 16-19W, any higher and I have to take really short pulls or I get a burnt wick (and a burnt tongue from juice popping onto it ^^). If I suddenly had to start setting the same coil to 10W or 25W on my pico with AF to get the same effect when it works properly at 16W on my VTC VTwo with AF, I would be wondering whats wrong/different with the pico. So... does that atomizer work at sane wattage levels in VW mode on pico? Clearer I hope.

buddermilch, 2017-02-16:
Yes it fires the same on both my mods running AF. 
TBXin thank you for your reply, I basically know how TC is working. I never locked the coil on original eleaf FW because it's not really locking anything, it's just not asking for a new coil. I did not assume AF to do the same silly thing, but I didn't really thought about it either :)

Anyway, I played a bit around with your suggestion of locking the coils resistance. And it seems if I lower the coils resistance it's working. But that makes sense as lowering the resistance does affect TC heavily. For example lowering it to 0.4 ohms instead of 0.6 TC will instantly get triggerd. 

So I assume there are some inconsistencies regarding my coil or atomizer. The difference between my rx2/3 and Pico in resistance reading are pretty huge, my Pico for example is reading the coil at 0.556 ohms, while the rx2/3 is reading it at 0.625 ohms. I will take a look at it this evening. 

But what I still don't really get is why the stock firmware provides me with an ok TC without locking the resistance. While I'm struggling with AF, are the algorithms so different ?

buddermilch, 2017-02-16:
Ok just a little update, so I put a new 26ga SS304 on my kayfun 5 and made the screws extra tight, and checked the 510 screw. 
I also reset the data and installed the FW again, and put an extra eye on the room temperature for the coil. Locked the resistance as suggested, and it's working flawlessly now. 

I don't know what was causing my problem exactly, but anyway it's working now and I'm happy with it :) 

Thank you guys for your help, I'm going to close the issue. 

Best regards
buddermilch

closed 2017-02-16 by buddermilch