maelstrom2001/ArcticFox #139: Protection off - Layout issues

Dil3mm4, 2017-02-17:
Hello. 
Pico Mega 80W here, having some probs with protection auto off after installing ArcticFox firmware.
I'm using a Cubis Pro with a SS316 (0.5), and I'm also using the same values that I was using with the stock firmware (28W, 250C°) and I keep getting the protection off when pressing fire button for more than 2 sec.
Having probs with the layouts too, modifying parameters in any view and uploading the new conf simply doesn't apply.
Thank you for your time spent on reading this.

TBXin, 2017-02-17:
Hello,

AF SS316 material is slightly different from SS316 Mode in the original firmware.
You can get experience like with original firmware if you will use TCR with value 120 on the ArcticFox or you can try [TFR] 316L material which is more suitable actually.

Best regards!

Dil3mm4, 2017-02-17:
Hello @TBXin, and thank you for your fast reply. 
If possible I'd like to understand why protection off gets triggered on standard SS316 mode.
At least to understand what's the issue with it. 
And still, any idea for the layout non-being applied issue? 

Thank you.

maelstrom2001, 2017-02-17:
@Dil3mm4 Standard SS316 modes (not TFR) have different TCR values in stock firmware and ArcticFox, 120 and 92 accordingly, so you'll have different coil temperature readings.
Concerning layouts, please make screenshots of your configurations (from NToolbox) that you can not upload to box, and steps to reproduce. Thanks.

Dil3mm4, 2017-02-17:
@maelstrom2001 is it possible to get the values from stock and then set 'em back on ArcticFox? 
About screenshots, they will be ready in 1hr, leaving work for lunch atm.

maelstrom2001, 2017-02-17:
This is equivalent of stock firmware:
[image]

Dil3mm4, 2017-02-17:
[image]
Can't modify this

Dil3mm4, 2017-02-17:
Dunno if the it's evaluated the SS316 for Pico Mega it's considered as an issue/bug. 
So I won't close this issue, if needed I hope that moderators will do that. 
I will open a new issue to ask something else. 
Thank you.

maelstrom2001, 2017-02-17:
For example:
[image]

closed 2017-02-18 by TBXin