maelstrom2001/ArcticFox #142: custom edit

obarone, 2017-02-17:
How can i edit the arctic fox sofware?
Thanks

TBXin, 2017-02-17:
Hello,

For what reason you want to edit ArcticFox firmware?

obarone, 2017-02-17:
I'm a student of IT and I want just to see if I can improve something inside the firmware and maybe commit it to the official one

TBXin, 2017-02-17:
I see. At this moment source code is not available for public.
We are not planning to make it public it at this point.
Anyway, you can submit your suggestions in form of issues and we can talk about them.

Best regards!

obarone, 2017-02-17:
My improvement are:
-Logo mode: fix logo on the screen, and return to the classic info screen only when press on +/- buttons
-Diaable clock(for energy) and disable the visible clock(like screensaver)
And, maybe a custom icon on the battery info(that's only because I use the percentage info, and the icon of the battery is useless)

TBXin, 2017-02-17:
> Diaable clock(for energy) 

Clock does not consume any extra energy until you are not turn on LSL mode.

> and disable the visible clock(like screensaver)

Just turn off screensaver and clock and you will never see them.

> And, maybe a custom icon on the battery info(that's only because I use the percentage info, and the icon of the battery is useless)

You can edit battery icon in the NFirmwareEditor.

closed 2017-02-18 by TBXin