maelstrom2001/ArcticFox #143: [ENHANCEMENT?] 16x96 UI

Dil3mm4, 2017-02-17:
Hello, 
it's possible to have a 16x96 UI ? 

Example attached: 
[image]
I'm not so good in Gimp, but that's the idea 😄 

Thank you for your time spent on reading this.

TBXin, 2017-02-18:
Hello,

Thanks for your suggestion, but i think this kind of layout is not convenient for almost all kind of data.
I mean that many counters have 3 or 4 digits.

Dil3mm4, 2017-02-18:
I know how to fix this if you are interested :)
Give me 30 mins I will do some example with GIMP :)

Dil3mm4, 2017-02-18:
But first, how many digits can we put in 16x96? Three?

TBXin, 2017-02-18:
Depending on the glyphs size. It's depend from the font that you will create for this kind of layout.

Dil3mm4, 2017-02-18:
There's a Max Value of digit tho? 
Even if it's an estimation.

TBXin, 2017-02-18:
Digits are from 0 to 9, but numbers can be very different, e.g. 250W, 0.30 Ohm, 0.302 Ohm, 23:49:52 and so on.

Dil3mm4, 2017-02-18:
So let's say 4 digits max per line? 
I need to know to generate a preview :P

TBXin, 2017-02-18:
4 digits and one dot, and if we are talking about current time - 6 digits and 2 colons.

Dil3mm4, 2017-02-18:
Example of 0.101 watt that won't fit in a 3 digits per line setting:
[image]

Dil3mm4, 2017-02-18:
If you like the concept I could illustrate how edit mode will work too.

vendeghlajos, 2017-02-18:
NICE!!!

Dil3mm4, 2017-02-20:
@TBXin did you saw the concepts ? :)

TBXin, 2017-02-24:
@Dil3mm4 yes :)

inabsentia-jacob, 2017-03-01:
How's this idea coming along? I was just looking at the clock on my iStick and thinking that I wouldn't mind it being vertical

Dil3mm4, 2017-03-04:
Do you think that it's a questionable layout for your next release @TBXin ?

StillVoidingWarranties, 2017-03-10:
I find nothing wrong with the landscape user interfaces. But I cannot think of any mod value you should need a precision higher than 3 significant digits so this should not stop anyone from doing a portrait-style UI. Microscopic fonts may, though.

Dil3mm4, 2017-03-11:
If no actual dev it's interested in generating a portrait UI, I would. 
But I'd need resources to do that, as I can see this project atm it's closed source. 
So, if some dev it's interested in this to happen, but has no time to do that, feel free to contact me, I'd do it with pleasure.

Dil3mm4, 2017-03-14:
@TBXin questionable layout huh? :) I'd like to have a reply even if no dev has time, I would be glad to help.

TBXin, 2017-03-14:
@Dil3mm4 kinda ^^ 

We can't quickly check the convenience of this interface, but at this point i have a strong understanding that it will be uncomfortable / unreadable.
That's is not your fault, it's just very tiny screen.

Anyway, you can create any sketch in 96x16 size and use it as a logo, just for checking how it will look on real device screen.

Also, we have almost no free space in the ROM - but space will be required for new glyphs, that are you designed above.

Dil3mm4, 2017-03-14:
@TBXin I know, but what's the point of generating a logo with that size ? 
I can't do new glyphs to try a real layout to see how it goes without a developer kit at least.

TBXin, 2017-03-14:
@Dil3mm4 Above you posted the sketches of the new interface. You can make them in the size 96х16 1bpp and upload to the device as a logo. So you can see how your sketch would look on a real device.

Dil3mm4, 2017-03-14:
@TBXin what happens if it does look good then? :)

TBXin, 2017-03-14:
@Dil3mm4 we will be able to upload it to yourself as well and say our point of view about this interface ^^

Dil3mm4, 2017-03-14:
@TBXin I didn't understand if you are saying that you will make another branch with that layout included if it looks good, sorry :D

TBXin, 2017-03-14:
@Dil3mm4 I say that we need to look at it first to say our opinion :)

Dil3mm4, 2017-03-14:
Okay @TBXin what you need then? A screenshot with NFE to see?

TBXin, 2017-03-14:
@Dil3mm4 yeah, or "logo".