maelstrom2001/ArcticFox #145: [ENHANCEMENT?] Last time puff

Dil3mm4, 2017-02-17:
Hello, 
it would be nice to have last time puff.
Something like "You puffed at 15:30" updating the var every min.
Thank you.

TBXin, 2017-02-18:
Hello,

We won't make any features related to the time, because mostly all devices can't measure time correctly, only if the LSL options is ON, but then you will get more battery consume.

Dil3mm4, 2017-02-18:
It will be non-precise even if you set an updatable variable based on other parameters? 
Like "Puffed 1 min ago"

TBXin, 2017-02-18:
What kind of parameters? :) Device can't know correct time if it does not calculate it every second ^^

Dil3mm4, 2017-02-18:
You are right, sorry.
I was trying to think something alternative, but nothing came up into my mind. 
Anyway it will be a neaty feature, at least to include, then letting the user choose if he wants to activate it and use it knowing that could be a little battery draining on the vape (as already done with Light Sleep Mode).