maelstrom2001/ArcticFox #147: Strange battery behaviour

Dil3mm4, 2017-02-17:
Hello, 
battery seems to discharge faster.
After 5 puffs flows from 80% to 75%. 
Any chance that battery stats affect this ?

TBXin, 2017-02-18:
> Any chance that battery stats affect this ?

Sure.

Dil3mm4, 2017-02-18:
@TBXin what's the suggested thing to do? (btw other 7 puffs 91%)

TBXin, 2017-02-18:
@Dil3mm4 
1. how old is yout battery?
2. What battery profile you are chosen in the settings?
3. you can measure voltage on the battery with a more precisely tool, like a voltmeter and see what is going on.

Dil3mm4, 2017-02-18:
1: 3 days old
2: VT5 profile (as per battery spec)
3: I don't have knowledge to read such a specific sheet

TBXin, 2017-02-18:
@Dil3mm4 i see, your problems will go away if you set your battery profile in the device menu to the "GEN".

Dil3mm4, 2017-02-18:
@TBXin All done. 
I will try a full discharge and then I will charge it back and see if works as expected.
I'll update the issue when I got some results.
Thank you for your interest.

Dil3mm4, 2017-02-20:
Works good, but I can't get why it doesn't with VT5 battery profile.

TBXin, 2017-02-28:
@Dil3mm4 because VTC5 profile is not so good in the our reality ^^

closed 2017-02-28 by TBXin