maelstrom2001/ArcticFox #149: Win 10, BSOD page fault in non paged area, HIDCLASS.SYS Wisemec RX2/3 4.13

kingsqueak, 2017-02-18:
I'm getting BSODs that won't allow the AF firmware to load.

I start ntoolbox.exe, click on firmware update, plug in the Wisemec RX2/3, it can read it, shows the device info. I click on update from file, browse to the .bin image, click on update and get a BSOD.

Latest version of ntoolbox, latest version of the firmware, from github.

The Wisemec is on latest stock firmware 4.13

Other HID devices work fine, the Wisemec updater still works, other USB devices work, Escribe for another mod still works.

I've swapped USB ports and cables, same results.

The Windows 10 error is 
Page fault in non paged area and it is in HIDCLASS.SYS

Ersad26, 2017-02-19:
Try to use a usb 2.0 port. When i connect my device to 3.0 ports i get the same bsod.

Sanaki, 2017-02-22:
> Try to use a usb 2.0 port. When i connect my device to 3.0 ports i get the same bsod.

For what it's worth, this isn't standard. I have an RX2/3 and use it exclusively on a USB3 port on Windows 10 (64-bit) with no issues.

closed 2017-05-09 by maelstrom2001