maelstrom2001/ArcticFox #150: What I'd like to see added

wulfgrim, 2017-02-18:
First off, awesome firmware, thank you so much for this. And taking the time to contunue to update/support this, I'm very excited to see what It could become.

Firstly, I'd like to see a click lock function added in the "controls tab" in the click drop down. - locks all input/fire buttons from  accidental press

I too would like a AM/PM clock, but I'd also like the function to disable the clock for devices like rx200, etc that don't have internal oscillators, there's not much of a point as everytime I change the batteries the clock resets to 00:00 01/01/1980 something. on my evic vtwo the clock works flawlessly.

I think it would also be nice to be able to add new TFR materials and edit their names , I can't seem to figure out how to do this, and to have the ability to import from a csv file - I haven't had much luck with TC using a duel coil setup on here, the device kept saying temp protect and cutting the output. I'll have to try a single coil later and report back, as that might've not been a TC issue.

On TC mode your software chooses the appropriate power, is that correct? So when I setup a profile, I can set the power to 200W to drive a kayfun coil at 5 watts and I shouldn't have any issues? if not an update to the tc algorithm would be nice. power setting shouldn't matter, and the device should not be trying to push whatever power level the user set.

again, thankyou <3

wulfgrim, 2017-02-18:
Oh, and I noticed the puff counter, and puff time maxes out at 99,999 puffs and 99,999 seconds (which is a mear 27 hours of puffing) so if there was a box to disable recording puffs to save memory usage that might be useful too.

out of curiosity, how many kb of rom do they put on these SOC's?

docdeath, 2017-02-20:
as for rx,s clocks  you can disable in the tool box and i use the clock and never been a problem to my self and why am and pm ? 24 hr is fine

forkoz, 2017-02-20:
In the US AM/PM is used more than 24hr time. The date being reversed is actually distracting though. For battery changes it should write the last date/time to memory and go from that instead of resetting to the beginning. Even if it was a little off its better than reset.

wulfgrim, 2017-02-21:
^ this, the date being backwards can be a little confusing. and While I don't mind the 24 hour clock. having an option for a 12 hr or am/pm would be nice. not sure it would be terribly difficult to implement.

doc, for some reason it didn't write it to memory, and I'm more so glad, I wouldn't want the clock wearing out the writes of the NAND flash inside the chip anyways. I just disabled it. no biggie.

wulfgrim, 2017-02-22:
Is it possible to show the clock on charging screen?

maelstrom2001, 2017-02-22:
@wulfgrim enable "Clock on main screen" option, and clock will be on charging screen too.

Sanaki, 2017-02-22:
> what I would really like to see is the ability to upload TFR/TCR csv files and create my own custom materials. that's something evolv has that this software suit lacks at the moment.

@wulfgrim ArcticFox supports this properly. I didn't see it at first either, but take a look at #51.

closed 2017-02-23 by wulfgrim