maelstrom2001/ArcticFox #153: ArcticFox translation

MoDuLah, 2017-02-20:
I've completed by 80% the translation of the software in Greek but few controls are not "flexible" based on lang. So many phrases/words get cut by that.
I've tried to put the shortest words possible still not enough.

TBXin, 2017-02-20:
If German colleagues were able to put your words, then you can ^^

MoDuLah, 2017-02-20:
I don't really understand your message. Please enlighten me.
What German have to do with Greek?

TBXin, 2017-02-20:
I mean that average german words length is higher than Greek and German colleagues done translation very well and all words are still in the limits.

MoDuLah, 2017-02-20:
English: Preheat (7), Delay (5)
German: vorheizen (9), Pause (5),
Greek: Προθέρμανση (11), Αναμονή(7)
Ehm. Also that average thing doesn't seems correct. Greek have a Guinness Record about longest word. 172 characters ;)

MoDuLah, 2017-02-20:
It's easier to set a button parameter to Autosize = true
Than chopping of Greek words.

westor7, 2017-02-20:
Because i have translate a lot of programs into the Greek language , believe that the Greek language translation characters require more space because in some words that are simple to translated into the German or into an other language into the Greek to be a valid translation into a simple word or phrase it need a lot of more characters, so really you need to do something with that because the @MoDuLah here has right, that's one reason that the most of programs are not translated into the Greek language because the spacing issue part that needs re-arrange.

Dil3mm4, 2017-02-21:
I could translate it into Italian, if needed.

MoDuLah, 2017-02-21:
This is the file Westor7 if you have any idea to improve it go ahead.
[EL.lpack.txt](https://github.com/maelstrom2001/ArcticFox/files/790780/EL.lpack.txt)

Schwinni77, 2017-02-22:
Invalid because the NFirmwareEditor repository is the right place for translations: https://github.com/TBXin/NFirmwareEditor

MoDuLah, 2017-02-22:
thanks @Schwinni77 i wish @TBXin was guiding me there in the first place.

closed 2017-02-22 by MoDuLah