maelstrom2001/ArcticFox #154: Internal resistance

andreivoinea93, 2017-02-20:
Having a way to set an internal resistance value in ArcticFox and NFE would be great and possibly make TC even better.

[Check how escribe does it](https://www.youtube.com/watch?v=j_fBo6cIk1M) if you don't really know what I'm talking about.

Ps: You're doing a great job ;)

andreivoinea93, 2017-02-22:
Just an update...
I tested the internal resistance using the device monitor and a copper wire. Without an atty on, I shorted the connector using the copper wire and the device monitor shows 0.022-0.025 ohms on my Evic Primo. That's quite a high value for TC, so it looks like this feature would be REALLY helpful.

maelstrom2001, 2017-02-22:
> device monitor shows 0.022-0.025 ohms

Do not assume that these measurements are accurate ;) Unfortunately, it's not DNA. More or less accurate results can be provided while firing, by measuring significant current in load, while in standby mode the results can be considered approximate.
The idea is good, but for the correct operation the software should be improved.

andreivoinea93, 2017-02-22:
> Unfortunately, it's not DNA

I had this talk with lots of fanboys already, not suggesting that you are a fanboy, in fact I really appreciate your work. But DNA devices seem to become the Apple of vaping, being considered  better for no apparent reason.

In fact, I believe it's a matter of software and maths only, that's why I use ArcticFox and I'm trying my best to help you guys make it even better.

> results can be considered approximate

The fact that the values stay somehow stable makes me think the hardware is capable of making quite accurate measurements. Do you have a DNA board at hand and can confirm that they use WAY better hardware? I don't and I would be really curious in a component comparation but to be honest, I highly doubt so. Of course, the stability of the values in the device monitor also depend a lot on the filters you are using, but by the speed at which they change, I don't think that's a reason good enough to make me think the hardware does not give good measurement. Also, due to the nature of the process itself, we don't really need REAL values. We more kind of need STABLE values. An offset between the real resistance and the calculated resistance only means an offset in the real temperature and the set temperature, but it does not affect the STABILITY of the process. Instability in measurement on the other hand would make temp control useless. Maths.

After all, both DNA and Joyetech use the M0 and I doubt any of them have external CANs and CNAs with better resolution than the M0 itself, so only the software becomes the difference, which I hope we can improve ^^

traviswparker, 2017-02-22:
Can't you adjust for this with the SHUNT expert setting?

andreivoinea93, 2017-02-22:
> Also, does internal resistance matter? It's a static offset to both the cold and hot atomizer readings.

As I said above, it does not matter to the overall stability of power adjusting, but it does matter in how far the set temperature is from the real temperature. If we see the set temperature just as a number and not it's physical property, then no, resistance offset does not matter.

DLC86, 2017-03-14:
Hi all, installed this firmware a few days ago and came here to make the same request, but it seems andreivoinea93 anticipated me.
I know that with the shunt parameter a similar result can be obtained but since this is a percentage it should be readjusted every time we change coil to be precise. It woud be great to have the possibility to set a fixed offset (in ohm) for the resistance to compensate mod's internal resistance, just like in Escribe.
I think the fact that our mods are not DNAs should be an incentive to reduce all types of errors :)