maelstrom2001/ArcticFox #156: PICO DUAL wont power on

ZeusBomb1997, 2017-02-21:
I have bought my eleaf Pico dual 200w kit with melo 3 mini i have user my box 1 day but today my box dont work. The box cant be powered in the display is Black (the box dont work this is not the stealth mode) i have try another battery 18650 but is the same and the battery is charge.. help me please

maelstrom2001, 2017-02-21:
Try to flash stock firmware. Links are at the bottom of [product page](http://www.eleafworld.com/pico-dual/).

ZeusBomb1997, 2017-02-21:
I'm already in stock firmware 
And the eleaf updater Can recognize my Pico dual but my Pico dont Power on

maelstrom2001, 2017-02-21:
Well, if it don't work on stock firmware, return it to seller, it seems to be a hardware failure.

ZeusBomb1997, 2017-02-21:
And if i have upgraded your firmware i have the garantee?

TBXin, 2017-02-21:
@ZeusBomb1997 you should read warranty card that should be provided with the box, i think this kind of information can be there.

ZeusBomb1997, 2017-02-21:
The firmware is not menzioned in the warranty 
And Sorry for my bad english

TBXin, 2017-02-21:
@ZeusBomb1997 no problem and good luck with your devices!

ZeusBomb1997, 2017-02-21:
Thank you @tbxin  @maelstrom2001

closed 2017-02-21 by ZeusBomb1997