maelstrom2001/ArcticFox #161: Pre-heat does not activate with the PI Regulator enabled.

Starbuck3733T, 2017-02-23:
316L tfr wire, any preheat mode, display does not show the preheat wattage nor does the preheat function activate.  PI is great, hate running with it off!

andreivoinea93, 2017-02-23:
That's how it's supposed to work. That's also why they made the % value at which it kicks in. Set it to 50% and you will have preheat until it reaches half the target temp and PI for the other half.

TBXin, 2017-02-23:
Hello,

Because of default range value :) 
Range means when PI-Regulator is start working.
0% - immediately after pressing Fire button.
100% - when temperature of the coil are equals or greater than user defined.

So, if you want some boost with preheat at the start you can move range slightly further from the start, eg: 20-30%

Starbuck3733T, 2017-02-23:
thank you !!! so once it's over x%, the tfr curve doesn't get used anymore?

TBXin, 2017-02-23:
TFR Curve used to calculate temperature from resistance, and if you are using TFR materials - TFR Curve will be used during time of all puffs.

andreivoinea93, 2017-02-23:
let say you set the PI at 50% and want 200*C.

From 0% (0-20*C) to 50% (100*C) it's going to do preheat based on percent, power or preheat curve, however you have it set.
From 50% (100*C) to 100% (200*C) will use the PI to get to the temperature you want and keep it there.

Starbuck3733T, 2017-02-23:
Okay, makes perfect sense.  And working great here, too.  thanks!

closed 2017-02-23 by Starbuck3733T