maelstrom2001/ArcticFox #165: Current leakage on iStick TC100W

Tiankay64, 2017-02-25:
Hello

Im running ArcticFox Build 170201 on both my iStick TC100W and my Wismec RX200S. It runs jus fine on the 200S without any Problems, but i noticed something potentially dangerous on my TC100W. If i leave the Mod alone for some time, something is draining the batteries. This wasnt hapenning with the original 120W Firmware so im sure its a software problem. The leakage doesnt stop after regular cutoff and it drained one of my battery sets low enough for my LUC V4 Charger to dont even recognize them anymore.

So i set up a test yesterday. At 0:00 i put in a set of partially charged batteries. 3.78V charge. After 1 hour, nothing changed. But today after waking up at around 12:00 it had dropped to 3.28, so it lost about 0.50V just doing nothing.

Batterys are LG HE2s (set to GEN) and BVO is at 0.00. I have none of the clock or "light sleep" options enabled, I tried to disable USB charging so i could log the draining batteries but it still charges them. So the only thing i can provide is my settings export file. [IStick TC100W Tiankay.zip](https://github.com/maelstrom2001/ArcticFox/files/801160/IStick.TC100W.Tiankay.zip)

Peace
TK

MoDuLah, 2017-02-25:
Do you have LSL on?

smok-e, 2017-02-26:
I never had this happening on my iStick TC100W. I'm using it with the firmeware 170214. But all updates I did never had this problem, maybe it is a hardware issue?

Tiankay64, 2017-02-26:
It could be, but like i said i doubt it because it wasnt happening with the original eleaf 1.12 Firmware before.

Im running a new test today. I flashed the 170222 nightly yesterday and i will put some fully charged batteries into the mod to document the drain hourly by hand if it still appears with this build. I will report back with more data then later this day or tomorrow morning.

Peace
TK

smok-e, 2017-02-26:
I would install the original eleaf firmware again and see if it doesn't have the issue. And then try again the arcticfox firmware.

Tiankay64, 2017-02-27:
Mysterious. Wasnt happening before, but now also happens with stock firmware. So it really seems to be hardware, Case closed then

peace
TK

closed 2017-02-27 by Tiankay64