maelstrom2001/ArcticFox #166: eVic Primo USB Charging - Voltage and Charging status way off

FrankBKK, 2017-02-25:
eVic Primo with AF170222 - after inserting fully charged batteries Info reports 4.20V, 4.16, set as generic, no BVO
 
After several puffs the batteries are at 97%, Batt1 4.07V, Batt2 4.12V 
now plugging in USB charger immediately changes battery status to 87% and then shows Bat1 at 4.19V, Bat2 at 4.00V

that seems a bit off ... ;-)

andreivoinea93, 2017-02-25:
As long as your voltages are close and balance charging happens, the percentages don't really matter. They are based on a curve when you set the battery type.

TBXin, 2017-02-28:
Hello,

It's a normal behavior ^^

1. Batteries drop their voltage under load;
2. 2+ Batteries devices from joyetech/wismec/eleaft read wrong voltage;
3. Connection to the USB provides a few additional voltage and power to the PCB;

closed 2017-03-01 by FrankBKK