maelstrom2001/ArcticFox #167: TCR Error

Alunduyn, 2017-02-26:
Last build seems to keep throwing at me a TCR error. I locked the coil cold resistance, and after half a second of firing it cuts power and prompts "TCR Error". I switched back to stock and it worked (badly, because it picked up the hot coil as base resistance, but worked).

Tried changing the base resistance to higher values (0.22 to 0.26 in 0.01 increments) with no results.

I connected the device and monitored it with the Device Monitor, to see that at the right cold resistance the temp graph would spike way above the temp limit I did set, and then it would cut current. With every other resistance, the temperature was just too low and changed too slightly.

Using SS316. Issue is present both with SS316 and [TFR]316.

Tried being as detailed as possible.

maelstrom2001, 2017-02-26:
You can try to decrement base resistance (cold locked/saved) or disable TCR check.
For example, if you have used heavy coil and box saved higher base resistance than it should be, while firing the difference between locked and real resistance will be too small, and it will show error message.

guglielmoTaro, 2017-02-26:
Me too have an error like this, really small coil btw... 1,6 ohm ss316 30Ga

sandbagjohnny, 2017-02-26:
@guglielmoTaro @Alunduyn - Out of curiosity, are these contact coils, or spaced coils?

guglielmoTaro, 2017-02-27:
.5mm spacing between the spires

guglielmoTaro, 2017-02-27:
With previous releases it was perfect, same coil on DNA works like a charm

Alunduyn, 2017-02-27:
@maelstrom2001 I forgot about TCR check. Unchecking it seems to solve the issue. Dumb me.

@sandbagjohnny Mine is as pressed as I could, it still has some minimal distance between the wraps. I don't have ceramic/insulated tools to squeeze the coil while hot. 8 wraps 24 AWG SS316 round coil, dual.

maelstrom2001, 2017-02-27:
@Alunduyn just for information - what power value you set for this build? We have an idea to rework TCR check procedure, need some statistics. Thank you :)

forkoz, 2017-02-27:
So we all have problems with TCR + SS?

andreivoinea93, 2017-02-27:
I had only with twisted 24GA or huge coils like that. Disabling the check fixed it though.

guglielmoTaro, 2017-02-28:
No difference between curve or value, only on SS

Sanaki, 2017-02-28:
For info only, I run a [pretty beefy](http://imgur.com/a/D6el7) 0.083 ohm titanium build running on the titanium TFR profile. If I lock when cold, I have no issues with TCR errors. Uploading from NFE Toolbox, reattaching the atomizer, or modifying the locked resistance manually will cause a TCR re-check, which will fail to cause enough measurable heating if the coil isn't cold. If you're certain you're using a TC-compatible coil, disabling the TCR check is going to be your best bet.

TCR check causes the wattage to freeze at a low point (40W when ramping to 200W, not certain if that value changes with asked wattage or given resistance) as it watches for resistance rise. If that reduced resistance can't cause the coil to heat beyond where it already is, it assumes you're using something like kanthal and don't realize it, which is when it errors. The link I posted near the top of this comment has that check visible in the graph at the bottom.

forkoz, 2017-03-01:
I think it fires at 30w always. So I vape at 10w and it fires 3x the power I want. I think its good if you switch between materials a lot but otherwise pointless. You would quickly see you're firing kanthal when the temperature never changes and power doesn't ramp down. The most useful feature of TCR check is you can turn it off.

sala91, 2017-04-09:
I have also problems with my RX300 and running coils in temp control. My setup is TFV12 with it. If I put this on my Minikin V2 it handles temp control like a champ. So far my hack for temp control has been to set resistance on higher than it is and hope for best. If I set it to real resistance, it never works, in additinal TCR check has to be turned off. The material used in SMOK TFV8 and TFV12 is SS316 so in theory, all of their tanks should work fine. I just wish the experience would be better - from starting of resistance reading to ending with actually working with bunch of premade coils. 

Getting RX300 with SMOK SS316 coils working would be good idea, as it is easy to test and make adjustments to software for it to work. ATM the experience is sub-par.

CasperCrasher, 2017-04-26:
same problem here.
i made 2 twisted coils (using 2 SS316L 26ga wire) 4 spires each ones on a 3mm spike.
Coils are read around 0,15ohm. Click fire and TCR Error appear.
Disabling TCR Check works BUT when i set Temp.Control on 260° the real time monitor read always 21° only!
So temperature control never intervenes!
Someone with the same problem?

druli, 2017-05-26:
same problem here... :( Wismec predator AF 170510