maelstrom2001/ArcticFox #169: Request: Less than 1 minute screen saver

GnFlala, 2017-02-26:
What the title says. 

Thanks for your work.

wulfgrim, 2017-02-28:
and longer than 5s of puff time. Thanks (:

GnFlala, 2017-04-30:
please add less than 1 min screen saver. i already have burn in issues on my screen.

ZenFree, 2017-05-01:
@GnFlala I've never heard of burn-in issues in ecig mods. If it's similar to TV burn-in then you need to display a reverse image to get rid of it. Just take an image that was burned-in, edit in photoshop to make it look like negative photo and try to display it on the mod until burn-in fades. Hope it helps.

GnFlala, 2017-05-01:
that is a solution that will indeed work man. really thanks for the idea.

BigBadBlackWolf, 2017-05-01:
With myevic there is the possibility to switch the screen invert on or off.
This should also be realized at Arctic Fox.

TBXin, 2017-05-01:
> Really?

And ML/Day counter too ^^

GnFlala, 2017-05-01:
i do not think that we should compare different firmwares. each have is advantages. I prefer this for the ease of use (menu is more easy to the way my mind works), but most of all because the devs work on it.

No i do not think that an invert screen is something we need. its my fault that i used a very high screen contrast, but it would be nice for others to have a warning about this.  I am still left with the problem that with the sun conditions in my country i cannot see the screen with low contrast but that is easy fixed if i do not use screensaver and having the screen turn off fast. 

And i really do not care what other firmwares have. If i did i would go there.