maelstrom2001/ArcticFox #170: Increase the relax delay of preheat function

westor7, 2017-02-26:
Hello,

Now as i can see the limitation of the max delay for relax into the preheat function is 20-25 seconds, is there a way to make it be max to 60-90 seconds value, because 25 max is too low now.

- Thanks!

ceinro, 2017-02-28:
He has a point ! Can be longer.

TBXin, 2017-02-28:
Hello @westor7 

Preheat delay timer restarts each time when your are making puff >=preheat time.
In other words, usually you should not press the Fire button for the entire time of the preheat delay to again obtain preheat.

westor7, 2017-03-01:
@TBXin Yes, but the head request here is to increase this delay value from 20-25 that is max now to 60-90.

TBXin, 2017-03-01:
@westor7 i'am trying to understand for what reason you need so long delay.

ceinro, 2017-03-01:
Because some coils dont cool down in 20s and preheating them will cause them to overheat as the initial temp is higher than cold temperature.

TBXin, 2017-03-01:
@ceinro above I described how the delay is works. You can get "overheating" only if you will make puffs every 20 sec (with cooldown 20).

And i am really confused, because i use coils with resistance around 0.1-0.2 ohms (tripple staggered / stapples and so on) and even they have no problems with cool down for 20 seconds.

westor7, 2017-03-01:
@TBXin i cannot see here why is an limitation about 25 only max seconds, why not at least 60 seconds (1min), i vape really very different way from you, and i really need a big relax at least 1 min in order to be useful for me the preheat function, i cannot see any negative action by increasing it from 25 to 60 for example.

- Thanks!

Sanaki, 2017-03-01:
For info only (I don't use preheat), my 0.083 ohm titanium coils take 65 seconds to return from 350F to within 15% (0.095 ohms) of base resistance and 135 seconds total to drop to 0.09 ohms (still warm, but not hot). While I personally have no skin in this game, I can fully understand why longer times would be needed with heavier coils.

westor7, 2017-03-08:
@TBXin So, will you increase the max delay limitation value from 25 to 60 or more ?

- Thanks!

TBXin, 2017-03-08:
Will be available in the next release.

om22, 2017-04-10:
@TBXin would it be possible to restart the counter even when puff length is under the set preheat time?

For example: I like to preheat @60W for 2-3 seconds only for the first puff so my clapton gets to temperature. After that I almost only use puffs under 2-3sec (at about 30-40w) so preheat will activate again even though I kept the coil warm with my small puffs.

maelstrom2001, 2017-05-08:
Released.

closed 2017-05-08 by maelstrom2001