maelstrom2001/ArcticFox #172: Request: extended Wake-up Time

Presskohle, 2017-02-28:
Hi
As a TC-user I was always annoyed that the chip goes into deep sleep after 3 minutes. If I pick up my mod after ~3-5min. to have a vape, the resistance gets reevaluated after waking up from deep sleep. Which means I get greeted with the same/new coil question or with "SMART" activated, with the Smart Profile Selector (which even logs in the new wrong/to high resistance).

3 Minutes isn't simply enough time to cool down the atomizer to its correct cold resistance. Would it be possible to extend the wake-up time to..let's say 10 minutes (which should be enough cooldown time)? I don't think it would be noticeable battery drain/negatively hit battery-runtime.

I know that by locking the resistance and turning off Smart Profiles would "solve" my problem, but the Smart feature is (for me) too good to be turned off. :)

best regards,

TBXin, 2017-02-28:
Hello,

> I know that by locking the resistance and turning off Smart Profiles would "solve" my problem

Hm, to calculate correct temperature you should always lock resistance.

> turning off Smart Profiles

Can you provide resistance values of cold and hot atty?
May be you just need to slightly increase smart range.

Presskohle, 2017-03-01:
Yes, I already played with the smart range, but the problem is: I use pretty similar builds between my atomizers (single coils ~0.55 Ohm; dual ~0.28) - if I increase the range too much, switched atomizers doesn't get recognized as "new".

maelstrom2001, 2017-04-07:
This is more relevant to the Smart algo than wake-up time. Anyway, Smart has been improved, and all fixes will be included in upcoming release :)