maelstrom2001/ArcticFox #174: LFP/IFR Support

Sanaki, 2017-02-28:
Is it possible (including on the hardware side) to support IFR (lithium iron phosphate) batteries? From what I can tell, minimum voltage can't currently be reduced any lower than 2.75v, whereas IFR batteries can discharge safely down to about 2.2v (starting from 3.6v rather than 4.2v). I realize this isn't something you want easily enabled, lest someone destroy their batteries attempting to be clever, but I'm curious if it's even possible. The vaping range wouldn't really be viable for these batteries otherwise.

maelstrom2001, 2017-03-01:
From MCU datasheet:

[image]

And it is only MCU requirements, but we have pretty much another parts on PCB...

Sanaki, 2017-03-01:
Good to know. Guess we'll have to see what comes of the IFRs. The 26650 can maintain above 2.75v excessively well, but the 18650 has some pretty severe dropoff at higher amperage.

[image]

[image]

wulfgrim, 2017-03-02:
while it'd be nice, I'm not sure it'd be such a great addition considering lower voltages = more current required to reach desired wattage, so it'll be straining the already tiny PCB even more.

Sanaki, 2017-03-02:
As maelstrom2001 said, lowering that voltage limit isn't reasonable. It'll depend entirely on newer cells with better voltage maintenance coming out. While it won't be fantastic for everyone (and currently wouldn't be viable for me), they pull notably higher amperage limits than expected. The 26650 apparently can take up to 75a safely, which is notable compared to the current offerings.

The main reason I'm interested is because legislators are threatening to restrict sales of anything they can vape-related, which is circling around to ICR/IMR/INR batteries due to their "volatility". The IFRs are significantly more stable, so if things go south, they may be what we're left with in some regions.

closed 2017-03-11 by Sanaki