maelstrom2001/ArcticFox #176: virtual cigarette

maddash1337, 2017-02-28:
I would like to have a virtual cigarette feature.
I am thinking about a option to limit the puffs for lets say a 10 minutes timeframe to 20 puffs. After 10 minutes the counter and timer should rest until the next puff. Also it would be nice if you can decide what timeframe and puff limit .

docdeath, 2017-02-28:
really  virtual cig  ? not being funny why vaping is to help give up smoking so why would you want to over complected vape mod with some thing that is  not needed   20 puffs in  10mins lol

maddash1337, 2017-02-28:
Because sometimes I join some smokers at work. And if I use my vape while they smoking I will vape more than I need. So it will reduce my amount of puffs.... At least I think so.

Dil3mm4, 2017-03-01:
I feel too @maddash1337 request, would be a nice feature to implement.

CCCheshire, 2017-03-03:
Totally agree! I desperately need this option too. Or little simplier - just possibility to set number of puffs to lock device. After locking puffs must be automatically set to 0 and after unlocking new vaping session can be started.

maddash1337, 2017-03-03:
Yeah would also be fine for me. Simpler - no timer required ... sounds good.

GomezAddams, 2017-03-06:
I agree with the **CCCheshire**'s formulation. It would be very nice if such option will be implemented.

Dil3mm4, 2017-03-08:
@TBXin @maelstrom2001 isn't this a questionable feature? :) 
Setting X number of puffs then lock (or a simple message) the device!

ilffej, 2017-03-12:
I think this is a good idea - maybe have an option to set auto shut down after a certain number/duration of puffs.

sala91, 2017-04-09:
I love this idea. Atm, vaping is like lesbian sex - you never know when to end...