maelstrom2001/ArcticFox #178: Request: Screen Contrast % Based on time

GnFlala, 2017-03-01:
Hello,

I really do not know if this is possible but since most mods have time maybe it is.

I would like to request automatic change of screen contrast based on time. 
2 time zones will be enough. One time zone for day where the screen contrast needs to be higher to be able to read the screen out under the sun, and a second time zone for night where highcontrast results the lcd screen being like a flashlight.

As a general rule i would say 
zone one 6 morning to 6 afternoon
and the 2ond zone from 6:01 in the afternoon to 5:59:59 in the morning.

Ofc selectable zones would be much better but i really do not know if all the above are possible.

Thanks in advance.

TBXin, 2017-03-01:
@GnFlala 

Thanks for your suggestion.

We are trying to make firmware which is providing persistent and convenient features to all supported devices.

Almost all devices does not have built-in RTC module which provides the correct clock rate. Accordingly, all functions that are related to time can not work properly. That's why we wont create any features related to the time.

Yes, you can turn on LSL or setup RTC-Speed in the device settings to get optimal clock rate, but many people will come here and create issues because their devices works as not they are expected.

GnFlala, 2017-03-01:
Indeed but cant we have it to show the settings only on devices with rtc or if you have lsl on or if you have clock working even with no x32 crystal?

I am in no way demanding things. I am just writing down my ideas. 
I am thankfull for what i get from this firmware and i really have no problem as it is. Maybe my idea work maybe not. That is not a problem really. Just happy devs reply.

So once again thanks.

sala91, 2017-04-09:
Since RX300 does not have clock cristal, a menu shotcut for brightness/contrast boost would be good for those moments you vape at the sun.

traviswparker, 2017-04-11:
My old DNA 40 has a good solution to this problem.

Normal screen brightness is about 50%, it jumps to 100% while firing, and dims to black when going to sleep.

I suggest a "Bright" as well as a "Dim" setting. If set, it jumps to 100% when pressing any key, after the time it dims back to normal.

TBXin, 2017-04-11:
@traviswparker I didn't have a dna40 and I unfortunately can't judge what was the screen but I have dna200 and I can confidently say that it is much brighter and more contrast than those screens that are installed in the devices from Joyetech/Wismec/Eleaf. The difference in brightness between 40% and 80% almost not. May be in the dark place that makes sense, but in the general case no.

sala91, 2017-04-11:
@TBXin the difference in screen brightness is huge in RX300 tho.

traviswparker, 2017-04-11:
@TBXin The DNA 40 did have a very bright screen at 100%. I could use it as a torch in a dark room. 

eVic/RX screens will vary with the model and maybe even between units. They source compatible screens from multiple suppliers and some are brighter.

GnFlala, 2017-04-11:
i still think that if either x32 (and crystal exists) or lsm is enabled by the user, auto changing brightness based on time is viable.

if x32 is on and crystal does not exist then automatically this feature can be disabled

TBXin, 2017-04-11:
@GnFlala at the moment we want to give all devices the same set of possibilities. Not to depend on the hardware limitations. 

Your idea deserves attention, but we are still developing in a different way. If we reconsider our approach in the future, we may do as you suggest.