maelstrom2001/ArcticFox #180: any tips on how to set clock right

docdeath, 2017-03-01:
any tips on setting clock  right with out low power state

Sanaki, 2017-03-01:
If you mean just setting it, the clock is auto-set any time you upload from NFE Toolbox. If you mean keeping it running accurately and your device's standby doesn't allow that, try enabling x32. If that doesn't work, light sleep mode is going to be required, which will cost a bit more battery.

docdeath, 2017-03-01:
thanks but know all that and rx300 dose not have x32 and i don't want to use light sleep mode  as its big drain on batts  is why i am asking any other ways like how to set  rtc speed . thanks for your info but been using  custom firmware for  long time as in myevic which has its own how to for clock

picostick, 2017-03-06:
I too am curious about the RTC-speed function. How does this work exactly? Can it work to keep my clock on-time without using LSL?

docdeath, 2017-03-07:
after playing round with rtc speed for few days i can get my mod to stay with in 1-5 mins  off right time still messing and seams to be on right track now  and LSL  is off to

closed 2017-03-08 by TBXin