maelstrom2001/ArcticFox #181: Setting to disable the Fox logo during boot up

Gristrosia, 2017-03-01:
Hi,

I couldn't find it after looking through the wiki/readme page and the settings of the firmware. Is there a setting to disable the ArcticFox logo that shows up when the device turns on?

Since there doesn't seem to be a way to lock the fire button on the VTwo mini, I click 5 times to lock my device to not fire it when it's in my pocket and, it's not like I don't like seeing the cute fox with the build number, but after seeing it 20 times in a day it gets pretty annoying.

Thanks :)

TBXin, 2017-03-01:
Hi,

Why do you look at the screen when you turn? :)

Anyway, at this point there are no option to disable boot logo, but you can use "workaround" and press Fire button again to immediately hide logo.

Gristrosia, 2017-03-01:
It's not just about seeing your logo... 

> Anyway, at this point there are no option to disable boot logo, but you can use "workaround" and press Fire button again to immediately hide logo.

The Evic VTwo mini doesn't have an option to lock the fire button besides clicking 5 times to turn the device off. With the Original firmware when you turn the turn the device on, it doesn't display anything special and start directly/is ready to vape, which makes it a great mod to carry around at work or during social meetings.  With the custom firmware, I saw the screen and just waited for it to disapear before starting to vape. I didn't know the device could fire while the logo was being displayed. I assumed it was some kind of forced wait screen, and I never tried.

 I'll try that I guess. :s

closed 2017-03-08 by TBXin