maelstrom2001/ArcticFox #182: How to set pi-reg

miasmi, 2017-03-01:
Hi!
In pico 75w (with ss316L, 5 round coil, 3mm diameter: 0.56omh) how I can set the pi-reg? I use P:2500, i:300, set to 0% (always active).
with 13W, 235°, TCR316L, it going from 220° to 242°

 Is it correct?

Sanaki, 2017-03-01:
Give #91 a read. Should help clarify a bit. It takes some definite work to get it where you want it, but that should put you on the right path.

miasmi, 2017-03-03:
THANK YOU VERY MUCH! :-)

miasmi, 2017-03-03:
In synthesis: p il hao much watt range change and i is how fast it change, no?
with ss316L 5 spires 3mm diameter, 0.57ohm o found p:300, i:300 and 0% (always active), 15W power it works very well.
it works better with 25W power, more stable temperaure and hot smoke early, but stay always in "protection". Is it ok?

Jsl-1, 2017-03-05:
Hi, you should read this post, theres some advice about power setting with PI: https://github.com/ClockSelect/myevic/issues/132
With à similar build 6 spire, 24awg ss316l 0,43ohm and same pico, i have better results at full power. I lower to 60 watt because device reports low battery too early with full power, because you need more voltage to fire at 75w. 
 I dont know if it's optimal, but I use p=700, i=300. Temp is reached almost instantly and quiet stable. Max 237° in monitor with 230° set on device for a 5 second puff.

sandbagjohnny, 2017-03-06:
Current coil: 6 wrap vertical coil.. 3mm spaced with 28ga ss. Device is a Pico with a subtank mini.

Settings:

* 316L TFR mode
* PI regulation ON.
* No preheat.
* Ramp up 75w.
* Range = 0%
* P = 300
* I = 300

[PI regulation with above settings](http://imgur.com/ax8sq2C)

[Stock algo, same ramp up as used in PI tests](http://imgur.com/UgCJuMH)

[Stock algo, ramp up @ 20 watts](http://imgur.com/dvUguhU)

*Temp control issues*

In most cases problems arise from these issues; Using the relaxed TCR checking and/or PI mode is reported to work well for:

* Bad TCR from mismatched multiple coils
* Bad TCR from multimetal coils

Things that will adversely effect TC performance that firmware won't fix:

* Transient shorts (SS doesn't work as well with contact coils vs spaced coils In TC mode. In Wattage it's not noticeable because a transient resistance spike doesn't effect the regulator in real time like it does in TC mode.)
* Build resistance too close to the edge of the acceptable allowed range for the wiretype.

closed 2017-03-08 by TBXin