maelstrom2001/ArcticFox #187: Resistance Reading Issue

roaminghawk, 2017-03-03:
Hi,

I have a problem resistance reading on my evic vtc mini AF v 170201.
When in TC mode i get this chart:
http://i.imgur.com/LC9RcTY.png
This is a dry coil, as you see in the chart, The resistance reading stopped in two places and then continued to work. This is with PI-Reg turned off.
Here is another example (same dry coil):
http://i.imgur.com/eYDEnqU.png
This happen also in normal working coils.
With PI-Reg turned on I don't see this gap in resistance reading.
Here is a chart with PI-Reg On Range 90% P,I at default values:
http://i.imgur.com/TfWSC9M.png
There is small dips in resisatnce reading, but not like the ones in the above charts.
Also in normal Wattage mode I get this chart:
http://i.imgur.com/a5kjujz.png
As you can see, there is a dip in the curve happening at random.

I have SHUNT at 94% because my evic reads resistances a little higher. I also tried with SHUNT set to 100% and there is no difference of the above results.
This correct value was compared with a new Pico mod and using an LCR meter reading resistance of an atomizer at 1 KHz.

Is my device defective or is there anything I can do.? Also if my device was the issue, what is the component in the device that might cause this issue?

Thanks.

TBXin, 2017-03-03:
Hello,

Probably issue with your attys connection or screws are not tightened.

roaminghawk, 2017-03-03:
Thank you for the reply.
Which screws do you mean? The attys screws or the mods internal screws? 
This problem happens with different attys all running TC coils. So far I've tried 4 different atomizers.

TBXin, 2017-03-03:
> Which screws do you mean?

attys screws, but if you are telling that issue reproduces with different attys... i guess problem with your Pico connector.

roaminghawk, 2017-03-03:
Thanks again. 
Which connector?

TBXin, 2017-03-03:
510

roaminghawk, 2017-03-03:
I checked the 510 connection on the board and the solder is firmly connected.

joeblowma, 2017-03-04:
I had a similar issue when using SS316L coils in TC mode on VTwo mini, took it apart like you hoping to find cold solder some where and nothing. As well, the problem went away when I put the ato on my pico. Official fw either masked or didn't have the problem, but myevic did. I put a line to live resistance and noted it would drop to 0.00 in the 'gaps', and this didn't occur in VW mode on the same mod/coil.

Since, I've cleaned the 510 (both mod and ato) with some ear swabs and isopropyl (just enough to clean, not reach the rubber in the mod's 510), made sure to tighten the mod down well (almost over tight, I'm worried about the threads lasting) when attaching, and started using spaced instead of unspaced SS coils I've not had an issue since (~2 months). Knock on wood, I guess.

I do have to say the VTwo's performance hasn't been as consistent as my pico in AF though. Would love to put a proper spring loaded 510 in there and see if that really is the problem, but no luck finding one that would actually fit in a mini box and I don't really have the tools to make a fat daddy or similar fit  ^^

roaminghawk, 2017-03-04:
Thank you for the detailed reply.
Glad I’m not the only one with the issue, it almost drove me mad yesterday.

The official firmware didn’t mask the issue, if you pay close attention to the resistance reading onscreen on the stock firmware, you will notice that it will read zero sometimes while firing.

I also cleaned the 510 with cotton swabs and alcohol, and the problem didn’t go away, and that coil in the graphs is a spaced SS coil.
I think maybe you enabled the PID-Reg which makes this problem a little harder to notice.

Something else I noticed, if you put your ear next the 510 while no atomizer connected you will hear a faint clicking sound, I search around and people say that’s normal. Now disable PID-Reg, put an atty with a TC coil inside and try to hear the sound while firing, and sure enough the clicking sound is there while firing because the mod is trying to continuously read the resistance to adjust the power delivery. In my case when I see that gap in the chart I also hear that the clicking has stopped during the time of that gap and then continued.

I was going to order a VTwo to replace my VTC, but since it has the same problem It might be a better idea to get the new Primo with the “upgraded” 510 connection.

joeblowma, 2017-03-05:
I didn't actually start using PI-Reg up until a couple weeks ago, but even turning it off now after a couple dozen pulls I've not seen the problem recur on screen or with the monitor.

Until I saw your post I've put it down to "something needed breaking in, or, it will eventually come back when I least expect it, or, perhaps I was just unlucky in the lotto with this unit and a component is near fail but working well enough for now". It did it before with both a serpent mini and a cheapo kayfun 5 clone, which was what made me look at the mod closer.

Sorry not much help, since I can no longer get this mod to reliably give the symptoms. My next step would have been looking into myevic to see if I could spot something off specific to this device vs pico, as well as asking here like you have. From an electrical standpoint the FW must be detecting either an open connection or a short, either which is too brief to cause any protection to kick in and disable firing (which seems to be a very odd/baffling edge case code wise unless there is an external device to the cpu that causes a wait-for-reply).

fwiw my initial look into primo shows they are still using the same 510 connector as pico/vtc/wismec and just a different barrel above it. A friend got it recently and says its good quality like his vtc mini, whatever that's worth. Not seeing any teardowns on google though to be sure.