maelstrom2001/ArcticFox #188: TC Resistance Not Staying Locked

adro21, 2017-03-03:
The resistance is not staying locked in TC mode. If I switch from a TC profile with a locked resistance to Power profile, and then back to the TC profile, the resistance becomes unlocked. Is this expected behavior? 

- Switched coils at room temperature
- "Smart" is turned off for all profiles

I switch coils rather frequently, do I have to manually lock the resistance manually for TC every time I switch to it? If so, how do I do that?

TBXin, 2017-03-03:
Hello,

What firmware version are you using?
I cannot repeat the behavior you're talking about.
Resistance stay locked after switching between VW and TC profiles.

adro21, 2017-03-03:
I'm the latest version, just installed about a week ago. 

At first what would happen was, I would switch coils and it would prompt me if I wanted to update or keep settings. I've tried both and it's resulted in the resistance unlocking.

But today, when I was switching coils from TC to a VW profile, and then back to the TC profile, the resistance unlocked. I can confirm before I made the initial switch from the TC to VW profile, that the resistance was locked.

TBXin, 2017-03-03:
> I'm on the latest version, just installed about a week ago.

Version please ^^

> "Smart" is turned off for all profiles
> I would switch coils and it would prompt me if I wanted to update or keep settings

If you really disabled smart you will not see this dialog, so, i believe that smart is still enabled.
And yes, this is an expected behavior that lock will be removed after reassigning profile.

adro21, 2017-03-03:
Version: 170201

Smart was initially enabled, perhaps that's why I was getting the prompt.
But I can confirm Smart is definitely now unchecked for all profiles. 

Assuming this is expected behavior with Smart disabled, how do I properly lock the resistance from the device when switching from a VW to a TC profile?

TBXin, 2017-03-03:
> But I can confirm Smart is definitely now unchecked for all profiles.
Smart is a common setting, it is not set for a particular profile.

> Assuming this is expected behavior with Smart disabled
I mean that unlocking resistance is a normal behavior when you are reassigning profile from one of the dialogs.

> how do I properly lock the resistance from the device when switching from a VW to a TC profile?
What resistance you have locked and what is the current resistance when switching?

adro21, 2017-03-03:
> Smart is a common setting, it is not set for a particular profile.

Ohhhh I see that now. I thought for some reason based on the UI that the "Smart" checkbox was related to each profile. So I can confirm it is OFF as a common setting. (The reason I have it off is because it was randomly switching to a different profile.)

> What resistance you have locked and what is the current resistance when switching?

I had resistance originally locked at .31 for the TC profile. Then switched to VW coil, then back to TC profile.
Just now I picked up the unit and hit the fire button once and got this: http://d.pr/i/uyb3
(I never know what to select when this comes up, because I didn't switch the coil since earlier)

TBXin, 2017-03-03:
> and got this

Sorry but i can't make out the numbers, too small picture.
New 0.605
Old 0.540 
?

Your coils are warm or cold? Resistance is locked or no?

adro21, 2017-03-03:
Oh I'm sorry, not sure why that screenshot came in such poor quality. You have it correct:
New: 0.605
Old 0.540

Coils are cold and resistance is unlocked. 
I know for sure it was locked last night on my TC profile when I went to sleep.
This afternoon, I switched out the TC coil for one on a VW profile. (I did not get the New/Old prompt).
Then I switched back to the TC coil (still cold, have not hit fire) and noticed the resistance was now unlocked.

TBXin, 2017-03-03:
I can imagine only one case when resistance can be unlocked automatically... very strange and very rare case...
Your current profile is VW and you current resistance is X, you are switching to the TC profile with locked resistance X*2+- and only in this case (when locked resistance is higher than current resistance about twice) resistance will be unlocked...

adro21, 2017-03-03:
I can reliably reproduce it now too. Steps to reproduce:

1) Coil is twisted in and cold. Set to TC profile.
Settings: http://d.pr/i/vyU2   Device Screen: http://d.pr/i/zFk4
It is set to 0.31 and locked. Everything looks good so far.

2) Switched to VW coil and manually set to respective VW profile that I setup.
Settings: http://d.pr/i/LIPO   Device Screen: http://d.pr/i/1r7P
It is set 0.55 and unlocked. That also appears to be correct.

3) Then, without firing or warming up the coil, I switched back to the TC supported coil and then manually set it to my TC profile. 
Device Screen: http://d.pr/i/znsQ
Notice now, how the resistance is set to 0.29 for some reason and it is also unlocked.

TBXin, 2017-03-03:
Why you have locked 0.31 resistance while your coil real resistance is 0.55?

adro21, 2017-03-03:
That's the value I originally locked my TC coil resistance to...
The 0.55 is the resistance for the VW coil...isn't it?

TBXin, 2017-03-03:
You are switching profiles or attys? :)
Real resistance should be same in all kind of profiles.

adro21, 2017-03-03:
I'm switching both coils and profiles. I manually twist off the coil first, then hit the fire button 3 times, then change the profile to match the coil I've just twisted on.

TBXin, 2017-03-03:
some dialogues appear like new/same coil?

adro21, 2017-03-03:
In the above steps to reproduce, I did not get any dialogues

TBXin, 2017-03-03:
Hm... very strange, need some time to investigate. I have no idea what happens with your device.

adro21, 2017-03-03:
It's a brand new device as well, have only owned it one week.

TBXin, 2017-03-03:
I was able to reproduce this behavior. It really looks like a bug. Will keep you posted.

sala91, 2017-04-09:
I can confirm, same issue with RX300

ceinro, 2017-04-11:
I have the same problem with pico and i have tried to explain it here but unfortunately i wasn't able to reproduce it. Glad someone else has ! Waiting for update !

nbraunstein, 2017-04-17:
Same issue on RX 2/3

maelstrom2001, 2017-05-08:
Released.

closed 2017-05-08 by maelstrom2001