maelstrom2001/ArcticFox #191: monitor and display max temperature in power mode

1881hc, 2017-03-04:
It would be nice if the firmware could display the max temperature the coil reaches on the last puff in watts mode.  temperature sensing would be sensing and measuring but limiting would be turned off.  This would help users evaluate the difference between power mode and temp control.

TBXin, 2017-03-04:
Hello,

This feature is a bit confusing since to calculate temperature of the coil firmware should know from what material coil is done.

In our firmware VW is exactly the same material as the rest, for example Nickel. So, this will confuse people, because they will have to specify two material: vw + something else...

1881hc, 2017-03-04:
I was making a suggestion. Perhaps the function would only work when a specific coil material is specified.  If non is specified the feature is not turned on? or there could be a third choice in the drop down where there is temp control, power mode, and perhaps a third power mode with temperature sensing?   Perhaps a way to do this that iis not confusing will come to mind.  In the mean time thanks for your excellent work on this firmware.  It improves my Pico and vtc mini significantly.

Is there any chance of an android version in the future?

snigel, 2017-03-05:
I would be interested in finding this out what temperatures I'm running. I never seem to get it right using TCR compared to fixed wattage, switching back and forth with the same TCR able coil.

1881hc, 2017-03-05:
snigel, I brought this up in another forum and it was pointed out that setting the temp very high in temp mode should be the same as operating in power mode.  in that situation temperature would be monitored but as long as you're not hitting 600 degrees tc mode should not influence the vape.

snigel, 2017-03-08:
@1881hc when I use temp mode at 315c (600f), the monitor shows me somewhere around ~20-30w, giving a dry vape experience compared to running 35-40w power mode.

Surely, I may have calibrated the coil resistance badly. Seeing the (incorrectly calibrated) temperature during power mode would help calibrating this back to temp mode.