maelstrom2001/ArcticFox #193: Option to hide "temp-protected"

wulfgrim, 2017-03-05:
That's pretty much it.. I'd like to see current temperature displayed on the screen instead of a message telling me TC is doing it's job..

maelstrom2001, 2017-03-05:
You can setup information line to show the last temperature, and you'll get what you want.

wulfgrim, 2017-03-05:
Thanks, I'll do that for the classic layout, but in the circle layout, there is no such option.

maelstrom2001, 2017-03-05:
[image]

wulfgrim, 2017-03-06:
Thankyou, So it wouldn't be possible to have the last tempture displayed where protection is shown in the future?
[image]

maelstrom2001, 2017-03-06:
If you set infolines as I provided on screenshot above, you'll see the coil temperature instead of output current, and it will be shown always while firing, even in protection state.

wulfgrim, 2017-03-06:
It's tiny and hard to read. and I apologize for nitpicking, but I find protection flashes annoying. keep up the awesome work. you're really appreciated in this community.

traviswparker, 2017-03-06:
Turn t.dom off and it will just be a tiny "max" below the wattage. You don't really need to know what the actual temperature is because AF keeps it right at the set temp. 

Having live wattage shown while firing is better anyway because you get a feeling for when the wattage is higher or lower than it should be at temperature... good way to know when you need to replace your wick or coil.

joeblowma, 2017-03-09:
I too find the protect message a little annoying (and have run into many places where a TC newbie thought it was an actual error message in official firmwares ^^). Being able to choose to disable it, or indicate it a different way in circle layout (there is a little room above still in the circle for a mark), without putting temp on a tiny line would be definitely be appreciated - even if not entirely necessary.

Plus, I've always thought the settings for "last" items were more "previous to current" as opposed to "live" ^^

wulfgrim, 2017-03-09:
glad I'm not the only one. please please add this feature. for what it's worth, escribe has the same option. while I appreciate your help, I can't ask you enough to actually implement this feature. I think it will greatly improve the aesthetics of using TC on your firmware and the device.

think of it as having a microwave that chimed the whole time it was on, instead of just when the timer would go off.

TBXin, 2017-03-12:
@wulfgrim if we will remove "temp. protected" tooltip you will see temperature that you set for current profile. I can't imagine why you want to this.

Temp. protection indicates that your coil has reached the preset temperature and the device maintains this temperature by reducing power.

wulfgrim, 2017-03-13:
So if you added a option to remove the tooltip the device wouldn't just show the current tempture in the same screen? or something similar to this? 
[image]

joeblowma, 2017-03-13:
Only thing I can think of is perhaps suspending current temp calc for the main display refresh frees up some cycles to do calcs to do control... it clearly goes over regularly, which tbh I can't really see without connecting it to a pc because of the 'protection' message. Anyway, I still love it (enough to have even spent a ridiculous amount of time making the font to my liking)
[image]
[image]

wulfgrim, 2017-03-13:
@joeblowma wouldn't think so. it does display current temp is last temp while firing afterall.. Yeah, i like the firmware too, and I'd love to see where it's going with developing and all, I'd like to see more customization in nfe toolbox as to what can be displayed/hidden.

RX300N, 2017-04-12:
Me personally found more informative to see the blinking instead of showing the wattage when puffing. Looking down to blinking display (when puffing) shows clearly and visibly that temperature reached the preset value. If there is too litlle vape you can adjust accordingly. If temeperature is reached very quickly with low vape production then mostly I am running low on liquid.  
If the display will show the wattage I cannot recognize quickly enough what does it show (rapid change) and therefore to make some action on my side. Trying it with no-puff - just fire butoon pressed  -is completely different situation with little relation to actual vaping IMO.