maelstrom2001/ArcticFox #194: Smok mod support

Maschine123, 2017-03-05:
Hi

I use your firmware on my iStick Pico and on my Evic vtwo and love the firmware. 
Now I bought a new Smok AL85 and want to flash ArcticFox on my Smok mod.
Are your working on firmware for this device? 

Thanks

TBXin, 2017-03-05:
Hello,

We are glad and very appreciate that you like our firmware.
But we are not going to create the firmware for Smok device.

Maschine123, 2017-03-05:
Hmm this is sad. 
Can I ask why?

TBXin, 2017-03-05:
We are not Smok fans ^^

Seriously speaking, imposible to port an existing firmware. Need to create a new one from scratch, and it is very complex and long task.

Maschine123, 2017-03-07:
So a firmware for smok devices will never happen?

TBXin, 2017-03-07:
more likely no than yes.

closed 2017-03-08 by TBXin