maelstrom2001/ArcticFox #196: Support for Eleaf Istick QC 200w

AAI688, 2017-03-06:
Eleaf Istick QC 200w has just arrived in stores near me. its basically the upgraded version of istick ipower 80w. can it get support from you guys? gonna buy it as soon as the support is there.

TBXin, 2017-03-06:
Hello,

We got plans to add support for this kind of devices, but i can't tell when it will be available.

adzam016, 2017-04-24:
Hey guys,

I too would be very happy to give a test run of the next Arctic Fox firmware on the Eleaf iStick QC 200W.

Please just give me a shout if you need bug testing etc. and I'll flash it onto my device. Currently have it installed on my iPower 80W so should be quite easy to see how it is working in comparison.

I have a little background in C++ if that helps.

Very much looking forward to the next build! Great work guys!

Thanks,

Adzam

maelstrom2001, 2017-05-08:
Released.

closed 2017-05-08 by maelstrom2001

adzam016, 2017-05-09:
SICK! Loving the new firmware guys! So far iStick QC is behaving itself as is my iPower.

Thank you so much!