maelstrom2001/ArcticFox #199: eVic Primo battery

zmsens55, 2017-03-06:
eVic Primo , af_170222.bin ,
I am using Samsung 25R battery.
Which profile should be applied? GEN? 25R?
When the 25R profiles were applied, 
the remaining battery charge moved freely.
However, it is stable when applying the GEN profile.

GnFlala, 2017-03-07:
25r profile will show more real precentage. 
Profiles do nothing more than make the % shown on screen follow the reallity. THey do not make battery last more or less or anything else.

TBXin, 2017-03-07:
Hello

In most cases i recommended to use GEN (Generic) profile.

The batteries are continuously being degraded, so the profile needs to be corrected approximately every weeks for optimal results. This is too way complex for regular user, and thats why battery profile selector located in the "Advanced" settings.

zmsens55, 2017-03-07:
After applying the 25r profile, charging is completed without reaching 100% when the battery is charged. On screen
95%.

GnFlala, 2017-03-07:
Do you charge the battery out of the mod? If yes maybe the battery has lost capacity.
If ypu charge it in the mod does your wall pack have the power to fully charge the battery? Does your mod?

I think i can understand why the general profile is better.

zmsens55, 2017-03-08:
Evic primo

zmsens55, 2017-03-08:
I charged on mod. I don't have battery charger

GnFlala, 2017-03-08:
Do jot expect to reach max then. Also each battery even if same model will have differences. All batteries degrade aftwr each use. All batteries behave differently. 
The profiles here do not affect battery but onlynthe visual presentage you see. 
It is normal when you disconect charger to see voltage drop. So if the profile has 100% at 4.20V and when you disconect it goes to 4.15 then on wcreen you will see less%.
Get a real charger or use the generic profile or tweak the profile so you can have the visual 100% be at the voltage your battery has when you disconect the charger. After a while when your battery degrades more ofc you will have the same problem. 
A no name battery or a non original/copycat battery will show these symptoms faster ofc

GnFlala, 2017-03-08:
Or just stop worrying

zmsens55, 2017-03-08:
How to use ohm lock function ???

GnFlala, 2017-03-08:
From menu (fire + up) go into the profile (1st menu item and get the ohm up or down even just 0.1. And exit

closed 2017-03-25 by maelstrom2001