maelstrom2001/ArcticFox #201: Battery drain

petermoffat, 2017-03-08:
I've just Arcticfox firmware on both my Evic Primo (nightly AF170214) and Cuboid Mini (stable AF 170201), and battery life on both has decreased dramatically, the Primo (2xLG HG2 3000mAh) going from 79% to 64% within a few puffs.

I'm running the same tank on both, a Geekvape Eagle with SS316, 0.26ohm single coil, temp at 220C, wattage at 55W, using a linear pre-heat with PI control, though the preheat and PI don't seem to affect this, battery drain is about the same using other curves with no PI control, and PI values, while making a big difference in the control of the device, don't appear to affect battery life and more or less?

maelstrom2001, 2017-03-08:
What battery profile you are set?

petermoffat, 2017-03-08:
It's set to LG HG2. Should I use another profile?

TBXin, 2017-03-08:
Try to use GEN profile.

petermoffat, 2017-03-08:
OK done, will test and see if it makes a difference

closed 2017-03-08 by TBXin