maelstrom2001/ArcticFox #202: Battery percentage calculated incorrectly?

snigel, 2017-03-08:
My Eleaf iPower is currently charging and is showing 36% at 3.71v.

That doesn't seem right to me, looking at this lithium ion table for example;
http://www.powerstream.com/lithium-ion-charge-voltage.htm

maelstrom2001, 2017-03-08:
Battery percentage calculation is based on battery discharge curve. Edit it to match your cell type.
Also, battery percentage is calculated when difference between previous % measuring and current exceeds 0,02 volt, so percentage refreshed is not quite frequently.

snigel, 2017-03-08:
The Eleaf iPower has an integrated battery. I'm guessing it's lithium ion. Haven't found any information on what specific battery model it is, haven't found anybody who has opened their unit.

When I check the ArcticFox Configuration, it's set as "Generic battery". I thought that was considered a generic lithium ion, no? What curve is the generic using?

Surely I could try to create a custom lithium ion curve for this unit, but if there are a "good enough" curve from another cell, I could pick that instead.

maelstrom2001, 2017-03-08:
> What curve is the generic using?

GEN curve for Li-Po cells using values provided by manufacturer in original firmware:
0 % - 3,40 V
10 % - 3,57 V
20 % - 3,67 V
30 % - 3,70 V
40 % - 3,73 V
50 % - 3,77 V
60 % - 3,82 V
70 % - 3,89 V
80 % - 3,95 V
90 % - 4,02 V
100 % - 4,10 V

snigel, 2017-03-08:
Thank you, 
I could try to improve the curve using a charger doctor and see how many mAh has been charged at different voltage levels.

Does the generic firmware for the eleaf iPower 80w say Li-Po?
Where can I find at what voltage the device shows low battery and when it stops charging?

maelstrom2001, 2017-03-08:
> Does the generic firmware for the eleaf iPower 80w say Li-Po?

The table above is contained in all original firmwares for boxes with built-in batteries.

> Where can I find at what voltage the device shows low battery

0% in NToolbox Discharge Profile Editor

> when it stops charging?

MCU can not control charging on single cell boxes. It just receive charging status from external IC which control charging process. There is no possibility to start or stop charging by firmware.

closed 2017-03-19 by TBXin