maelstrom2001/ArcticFox #204: eleaf pico not locking keys

archphile, 2017-03-08:
Hi all,

I have been using pico with latest arcticfox and it does not lock the keys when - and + are pressed together. It shows that the keys are locked but is still can press the power button and vape like it's unlocked.

Regards,

Michael

TBXin, 2017-03-08:
Hello,

Key lock has exactly the same behavior as in the original firmware.
Key lock blocks increment/decrement of the power/temperature (+- button), not fire button.

closed 2017-03-08 by TBXin

archphile, 2017-03-08:
I am sorry. I ve never used pico without arcticfox and I did not know it.

However it would be really useful if power button was locked too so that it would be easy to put it in the pocket etc.

Thanks,
Michael

TBXin, 2017-03-08:
@archphile 

No problem,

You can press fire button 5 times to turn on/off device.